ArduPilot/ardupilot #23580: 4.5 parameter rename tracking list

Hwurzburg, 2023-04-23:
As we are considering renaming many params in 4.5, I thought I would start a tracking list:
- [ ] LIM_ROLL_CD, LIM_PITCH_CD
- [ ] DSPOILR/DSPOILER
- [ ] STAB_PITCH_DOWN_CD
- [ ] TRIM_PITCH_CD
- [ ] LAND_PITCH_CD
- [ ] Consolidate LAND and TECS_LAND params

feel free to add to the list

tridge, 2023-04-23:
i planned on creating the list by finding all params with units of cm, cm/s, cd and cd/s to start with

tridge, 2023-04-23:
a --units option to param_parse.py to produce docs only for the listed unit types would be a big help

tridge, 2023-04-23:
we also need methods like: param_convert_cm(my_param) to auto-convert

IamPete1, 2023-04-23:
I would like to do the `COMPASS` -> `MAG` as part of the compass move to AP_Vehicle.

`THR_FAILSAFE` on plane to `RC_FAILSAFE`

timtuxworth, 2023-04-23:
Percent vs Pcnt vs PERCNT - how about no abbreviation? Does it add value?
LAND_FLAP_PERCNT: Landing flap percentage
TKOFF_FLAP_PCNT: Takeoff flap percentage
FLAP_1_PERCNT: Flap 1 percentage
FLAP_2_PERCNT: Flap 2 percentage
ARSPD_OFF_PCNT: Maximum offset cal speed error

timtuxworth, 2023-04-23:
Use a fixed number of digits for SERVO and RC numbering and put in a separator to make parsing easier. Fixes the current UI issue where the servo numbers in lists like the full parameter list or search results are out of order.
SERVO_01, SERVO_02 ... or even SERVO_001, SERVO_002. (and if the characters are really an issue use SRV_001 etc )
RC_001, RC_002 ... 
SERIAL_001 (or PORT_001 - this would be even better see below)

timtuxworth, 2023-04-23:
> * [ ]  Consolidate LAND and TECS_LAND params

As per my notes here ( https://discuss.ardupilot.org/t/suggestions-for-improvements-to-parameters-in-plane-4-4/99165/2 ), consider removing TECS from all the TECS parameters. It adds zero value to end users and only causes confusion to have TECS in the parameter name.

timtuxworth, 2023-04-24:
standardize various spellings of "pitch" - ideally no abbreviation, PITCH is good.
STAB_PITCH_DOWN: Low throttle pitch down trim
LIM_PITCH_MAX: Maximum Pitch Angle
LIM_PITCH_MIN: Minimum Pitch Angle
MAN_EXPO_PITCH: Manual input expo for pitch
CUST_ROT1_PITCH: Custom pitch
CUST_ROT2_PITCH: Custom pitch
RCMAP_PITCH: Pitch channel
TKOFF_LVL_PITCH: Takeoff mode altitude initial pitch
TKOFF_GND_PITCH: Takeoff run pitch demand
ACRO_PITCH_RATE: ACRO mode pitch rate
TRIM_PITCH_CD: Pitch angle offset
LAND_PITCH_CD: Landing Pitch
MNT1_PITCH_MIN: Mount Pitch angle minimum
MNT1_PITCH_MAX: Mount Pitch angle maximum
MNT2_PITCH_MIN: Mount Pitch angle minimum
MNT2_PITCH_MAX: Mount Pitch angle maximum
MNT2_LEAD_PTCH: Mount Pitch stabilization lead time
OSD[12345]_PITCH_EN: PITCH_EN
OSD[12345]_PITCH_X: PITCH_X
OSD[11235]_PITCH_Y: PITCH_Y
Q_TRIM_PITCH: Quadplane AHRS trim pitch
TECS_PITCH_MAX: Maximum pitch in auto flight
TECS_PITCH_MIN: Minimum pitch in auto flight
STAB_PITCH_DOWN: Low throttle pitch down trim
KFF_THR2PTCH: Throttle to Pitch Mix
PTCH2SRV_TCONST: Pitch Time Constant
PTCH2SRV_RMAX_UP: Pitch up max rate
PTCH2SRV_RMAX_DN: Pitch down max rate
PTCH2SRV_RLL: Roll compensation
PTCH_RATE_P: Pitch axis rate controller P gain
PTCH_RATE_I: Pitch axis rate controller I gain
PTCH_RATE_IMAX: Pitch axis rate controller I gain maximum
PTCH_RATE_D: Pitch axis rate controller D gain
PTCH_RATE_FF: Pitch axis rate controller feed forward
PTCH_RATE_FLTT: Pitch axis rate controller target frequency in Hz
PTCH_RATE_FLTE: Pitch axis rate controller error frequency in Hz
PTCH_RATE_FLTD: Pitch axis rate controller derivative frequency in Hz
PTCH_RATE_SMAX: Pitch slew rate limit

timtuxworth, 2023-04-24:
_OPTION vs _OPTIONS vs OPTS vs OPT
mostly we use _OPTIONS but there is 
CAN_Dn_UC_OPTION and 
DSPOILER_OPTS and 
SCR_DEBUG_OPTS and 
INS_HNTCn_OPTS and
INS_LOG_BAT_OPT 
oh and RCn_OPTION

amilcarlucas, 2023-04-24:
@timtuxworth I'm sure there is a limit on how long the parameter names can be.
That is the reason for mixing non abreviations and abreviations, to keep the parameter length under a limit.

timtuxworth, 2023-04-24:
> @timtuxworth I'm sure there is a limit on how long the parameter names can be.
> That is the reason for mixing non abreviations and abreviations, to keep the parameter length under a limit.

Then it would be better to choose a standard abbreviation and use it consistently in all cases.

timtuxworth, 2023-04-24:
Inconsistent use of ROLL and RLL. I'd vote for ROLL but consistency is so important that I'd say RLL in all cases if the alternative was to have a mix of ROLL andRLL.

timtuxworth, 2023-04-24:
Use of THROTTLE, THR and TH for throttle. IMO standardizing on THR would provide the best balance of meaning, #characters and consistency.

timtuxworth, 2023-04-25:
TECS_LAND_SPDWGT vs TECS_SPDWEIGHT - should at least be consistent, but it's actually Speed vs Pitch weighting. Also "SPD" is wrong because it's airspeed so should be ARSPD. IMO having TECS at the front adds no value, so how about:
PTCH_ARSPD_WGT
LAND_PTCH_ARSPD_WGT
I'm suggesting this way around because there are a number of other "pitch" parameters and this will get PTCH_SPD_WGT to display with them, but the same argument could be made about airspeed (hence my suggestion about a grouping or tagging mechanism.
ARSPD_PTCH_WGT
LAND_ARSPD_PTCH_WGT

timtuxworth, 2023-04-25:
> > > @timtuxworth I'm sure there is a limit on how long the parameter names can be.
> > > That is the reason for mixing non abreviations and abreviations, to keep the parameter length under a limit.
> > 
> > 
> > Then it would be better to choose a standard abbreviation and use it consistently in all cases.
> 
> Disagree. The abbreviations are forced on us by the character limit. If we follow this logic then all motor parameters should just be M_ since that is what quadplane does - but I think we can agree that this would be a retrograde step :)

So fix the core problem (as per my comment above). Your suggestion is just layering one bandaid on another.

timtuxworth, 2023-04-26:
Oh here's a good one, some parameters that use centidegrees that don't even have _CD in the name!!!
LIM_PITCH_MIN
LIM_PITCH_MAX
FLOW_ORIENT_YAW
Q_ANGLE_MAX
Q_A_SLEW_YAW
STEER2SRV_IMAX
STEER2SRV_DRTMIN

amilcarlucas, 2023-04-26:
The units are in the extra metadata of the parameters, so having units in the parameter name is both duplication and redundant.

timtuxworth, 2023-04-26:
> The units are in the extra metadata of the parameters, so having units in the parameter name is both duplication and redundant.

I understood one of the goals of this exercise was to eliminate the use of centidegrees (and centimeters) - was I wrong about that?

I'm suggesting all these as candidates to be converted to degrees.

timtuxworth, 2023-04-30:
TECS_FLARE_HGT and LAND_FLARE_ALT, not only the confusion between using "HGT" and "ALT" for basically the same thing (height above ground), but also these two parameters seem to be doing similar things.

Hwurzburg, 2023-04-30:
> TECS_FLARE_HGT and LAND_FLARE_ALT, not only the confusion between using "HGT" and "ALT" for basically the same thing (height above ground), but also these two parameters seem to be doing similar things.
> 
> There are a large number of parameters that use HGT for "height above ground" - but also many "ALT" parameters. Does this make sense? Should it not just be ALT all the way down?

they are different parms and control different parts of the flare

timtuxworth, 2023-05-03:
BATT_VLT_OFFSET (whereas all other BATT_VOLT parameters are not abbreviated)

Hwurzburg, 2023-05-03:
> BATT_VLT_OFFSET (whereas all other BATT_VOLT parameters are not abbreviated)

16 char limit....BATT2_VLT_OFFSET....

timtuxworth, 2023-05-05:
> > BATT_VLT_OFFSET (whereas all other BATT_VOLT parameters are not abbreviated)
> 
> 16 char limit....BATT2_VLT_OFFSET....

Best practice when abbreviating is to drop characters from the right (least significant first). This would be better to do by abbreviating "OFFSET" so that VOLT can be consistent. Good options are probably:

BATT_VOLT_OFFSET or
BATT_VOLT_OFS (seems to be commonly used)

If we do this BATT[0-9]*_VOLT* will match all relevant values, and be much easier for users searching to find all the values they are looking for.

timtuxworth, 2023-08-18:
I'm not sure if this belongs here, but it just threw me off on a new build. All the RC channel functions are set by RCx_OPTION _except_ 2 special cases RCMAP_ROLL and RCMAP_PITCH. I'm guessing this is historical, but is this time to fix this? Could the Roll and Yaw channel not just be set using RCx_OPTION (x = 1 and 4 by default) just like any other channel?

Hwurzburg, 2023-08-18:
@timtuxworth not true...and RCMAP(ie axis inputs) are not ANY RCx_OPTIONS at this time.. but  201=Roll 202=Pitch 203=Throttle 204=Yaw are reserved in the code...eventually the RCMAP function will be removed and those options  activated and  used..

timtuxworth, 2023-08-19:
> @timtuxworth not true...and RCMAP(ie axis inputs) are not ANY RCx_OPTIONS at this time.. but 201=Roll 202=Pitch 203=Throttle 204=Yaw are reserved in the code...eventually the RCMAP function will be removed and those options activated and used..

Great. So I'm saying maybe its time to implement 201=Roll 202=Pitch 203=Throttle 204=Yaw and drop RCMAP_ROLL and RCMAP_YAW.

tridge, 2023-12-19:
for airspeed I want:
 - AIRSPEED_CRUISE
 - AIRSPEED_MIN
 - AIRSPEED_MAX

for TRIM_PITCH_CD I want TRIM_PITCH_DEG

peterbarker, 2025-08-12:
I think we can do without this issue now, @Hwurzburg .... closing this one, but feel free to reopen if you disagree :-)

closed 2025-08-12 by peterbarker