ArduPilot/ardupilot #23613: Sign error in ellipsoid height calculation

makeller1, 2023-04-26:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**
Seeing a sign error on how the ellipsoid height is calculated: 
[The undulation is subtracted from alt here ](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.cpp#L1421)
Which doesn't make sense considering the alt is calculated as [height - undulation](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_SBF.cpp#L418)

That would imply ellipsoid height = height - 2 * undulation. 

**Version**
Master

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2023-04-27:
This is probably one for @tridge, @hendjoshsr71 and/or @BluemarkInnovations to look into as the code was added in PR https://github.com/ArduPilot/ardupilot/pull/21075

makeller1, 2023-05-31:
Addressed by PR https://github.com/ArduPilot/ardupilot/pull/23942.

tridge, 2023-06-06:
thanks! fixed with your PR

closed 2023-06-06 by tridge