ArduPilot/ardupilot #23617: LAND_PITCH_CD minimum touchdown is not being adhered to in TECS

Hwurzburg, 2023-04-26:
still struggling to get a touchdown in autoland that is not nose wheel first....I think LAND_PITCH_CD is being largely ignored as a min pitch at touchdown...I have been steadily raising it from default of 0 with no noticeable ffect in landings...at 3deg now
...its used in some calculations in update pitch, but not as a limit and therefore fairly ineffective....see this typical landing.....sink rate is transitioned as expected, but the minmum pitch TEC.pmin and TEC2.pmin are NOT +1rads (3deg) at projected touchdown and the DesPitch is still negative..even as it rose it was not enforced on DesPitch....this is not an issue that the CG is too nose heavy, the demand is not being applied as advertized in the LAND_PITCH_CD parameter... 

[image]

log:https://www.dropbox.com/s/77gofyg81igzn1c/2023-04-24%2014-00-13.bin?dl=0

the LAND_PITCH_CD should override (interpolated to be full effect at touchdown) any sink rate or airspeed pitch considerations as a min pitch demand in the flare  or its rather useless

timtuxworth, 2023-04-27:
Yes I noticed something similar when doing some SITL testing yesterday. LAND_PITCH_CD doesn't do anything.
I'd reinforce the nose wheel mount on that AeroScout - I'm surprised it has lasted this long!

Hwurzburg, 2023-04-27:
After playing with SITL a bit I realized that TECs does adhere to that limit, but does not linearly interpolate to get to it, instead it holds at~0 deg for most of the flare and only at the very end will it increase to the LAND_PITCH_CD....so if your sink rate prior to that is not tightly adhereing to the desired sink rate, you hit the ground too soon before that final pitch demand kicks in....

so it could indeed be a nose heavy issue such that the pitch PID loop cannot get the desired sink rate and you land short before the final flare where the LAND_PITCH_CD is demanded....I will try making the plane less nose heavy, but its within the manufacturers recommended range...since the flare is essentially a two part deal, we should add a parameter to adjust when the transition to the LAND_PITCH_CD occurs so that slightly nose heavy planes can begin the pitch demand transition to it sooner if need be...just increasing its value does not have much impact...it needs to start phasing it in sooner...

Hwurzburg, 2023-04-30:
by setting FLAR_HGT above LAND_FLARE_ALT it forces the LAND_PITCH_CD to active at the start of the flare with the demanded pitch slewing to it....will add info to wik

closed 2023-04-30 by Hwurzburg

timtuxworth, 2023-04-30:
Do you mean TECS_FLARE_HGT?