ArduPilot/ardupilot #23635: Do not ignore channel 17 and 18 from mavlink_rc_channels_override_t

Consti10, 2023-04-28:
## Feature request

For some unclear reason, ardupilot does not process the channels nr. 17 and 18 from the mavlink_rc_channels_override_t mavlink message.

Implementation:
https://github.com/ArduPilot/ardupilot/blob/feaed97601598220e05043fd86bde6c02bc1a212/libraries/GCS_MAVLink/GCS_Common.cpp#L3648

packet.chan17_raw and packet.chan18_raw (valid values in the mavlink message itself) are ignored.

Is there any substantial reason for that ? Or is it simply an oversight that has not been noticed until now ?

IamPete1, 2023-04-29:
We only support upto 16 RC input channels.

muramura, 2023-04-29:
I PR to add CHAN17 and CHAN18 to the table.
This will match the MAVLINK message.
I think 1-16 is fine for internal use.

Consti10, 2023-05-03:
I looked a bit more into the source code - Looks like https://github.com/ArduPilot/ardupilot/blob/95354ac5726a000d94f28475601641fda5304df3/libraries/RC_Channel/RC_Channel.h#L14

NUM_RC_CHANNELS is 16. Looks like what @IamPete1 said is correct, ardupilot only supports up to 16 channels and therefore it is probably best to keep ignoring channel 17 and 18. It could be documented though ;)