ArduPilot/ardupilot #23664: install-prereqs-ubuntu.sh fails for ubuntu 23.04

AndKe, 2023-05-02:
## Bug report

**Issue details**

install-prereqs-ubuntu.sh needs changes for python 

**Version**
master

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
NA

**Hardware type**
NA

**Logs**

```
---------- ./install-prereqs-ubuntu.sh start ----------
+ '[' 1000 == 0 ']'
+ OPT=/opt
+ ARDUPILOT_TOOLS=Tools/autotest
+ ASSUME_YES=false
+ QUIET=false
+ sep='##############################################'
+ OPTIND=1
+ getopts yq opt
+ APT_GET='sudo apt-get'
+ false
+ false
+ sudo apt-get update
[sudo] password for andre: 
Ign:1 https://repo.vivaldi.com/stable/deb stable InRelease
Hit:2 https://ocean.surfshark.com/debian stretch InRelease                                         
Hit:3 https://repo.vivaldi.com/stable/deb stable Release                                           
Get:4 http://no.archive.ubuntu.com/ubuntu lunar InRelease [267 kB]                                 
Get:6 http://security.ubuntu.com/ubuntu lunar-security InRelease [90,7 kB]
Get:7 http://no.archive.ubuntu.com/ubuntu lunar-updates InRelease [90,7 kB]
Hit:8 http://no.archive.ubuntu.com/ubuntu lunar-backports InRelease   
Fetched 448 kB in 1s (542 kB/s)                           
Reading package lists... Done
+ package_is_installed lsb-release
+ dpkg-query -W '-f=${Status}' lsb-release
+ grep -c 'ok installed'
1
++ lsb_release -c -s
+ RELEASE_CODENAME=lunar
+ case ${RELEASE_CODENAME} in
+ PYTHON_V=python3
+ PIP=pip3
+ '[' lunar == bionic ']'
+ '[' lunar == buster ']'
+ '[' lunar == focal ']'
+ '[' lunar == jammy ']'
+ '[' lunar == groovy ']'
+ '[' lunar == bullseye ']'
++ apt-cache search -n '^libcsfml-audio'
++ cut '-d ' -f1
++ head -1
++ grep -Eo '[+-]?[0-9]+([.][0-9]+)?'
+ SITLCFML_VERSION=2.5
++ apt-cache search -n '^libsfml-audio'
[... 117 lines cut ...]
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
+ '[' lunar == bionic ']'
+ pip3 install --user -U future lxml pymavlink MAVProxy pexpect flake8 geocoder empy dronecan pygame intelhex
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

```

amilcarlucas, 2023-05-03:
can you try today?

AndKe, 2023-05-03:
Hi @amilcarlucas 
I just tried:

please consider mentioning cons/pros for this option :
"Make ArduPilot venv default for python [N/y]?"
I went with the default. :)  

I got a few warnings like:
```
  DEPRECATION: empy is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559

```

and in the case of empy  and future - it seems that the best way is to do "sudo apt install python3-empy"   
The other warnings were about wxPython and MAVProxy.

The script was completed successfully.
Thank you.

peterbarker, 2023-05-03:
Thanks @amilcarlucas , @AndKe

closed 2023-05-03 by peterbarker