ArduPilot/ardupilot #23691: AP_MotorsHeli Rework Roadmap

Gone4Dirt, 2023-05-05:
Through recent use and digging through AP_MotorsHeli it is evident that there are a number of issues that need addressing and number of features that could make user setup a lot easier.  This Issue defines a list of all issues and impovements and provides a means to keep track of what is going on and when.  Please add to this as realted issues are found.

**Rework Required**
_These are reworks that will bring heli motors more into line with conventions of the rest of the code base.  These do not change functionaility._
- [ ] AP_Motors example tool to test equivlance of outputs to allow for easier checking of rework PRs. (#23656)
- [ ] Remove unused functions and uneccassary virtuals. (#23667)
- [ ] Move all swashplate based calculations into AP_MotorsHeli_Swash.  We are currently doing a mix of swash calculations in the frame-specific code and swashplate.  Avoid duplication of code, improve maintainability and readbility. (#23684)
- [ ] Make calculation of roll-pitch-yaw-collective factors generic, and make presets go via the generic case.  One code path to maintain.
- [ ] Dual heli does not correctly limit cyclic angle in both axis.  Currently, you have one unlimited control axis and you have to set H_CYC_MAX on the other axis so that it matches the unlimited one.  For example, you cannot reduce cyclic angle and have a symetric swash plate.

**Feature Improvements**
_These are improvements that we really ought to be doing but not currently_
- [ ] Fix control saturation/output limiting flags.  There are a number of cases where we are not flagging control saturation at all or we are not doing it correctly.
- [ ] Impove logging of limit flags
- [ ] Fix use of get_motor_mask() to not include swashplate servos.

**New Features**
_These add new functionaility_
- [ ] Add swashplate logging to make it easier for users to find appropriate log data e.g. collective and particularly collective 2,3,4 for multi head frames. (#23655)
- [ ] Allow use of SERVO*_MIN _TRIM _MAX functions and blade angle linerisation curves to make swashplate setup easier and more accurate.

bnsgeyer, 2023-05-05:
@IamPete1 @Gone4Dirt a lot of this is great stuff and completely necessary. I really appreciate the help. I have one comment and one concern. 
I want to comment on rework required item regarding the limit cyclic angle in both axes on dual heli’s. Longitudinal cyclic is not limited because it is not used for control of a dual Heli. I agree that it should be included because I intend to implement longitudinal cyclic control in the future.  

My concern is with the future enhancements item regarding the use of the servo min and max for swashplate set up. Be sure you understand how the swashplate Servo output is determined and why we fix the min and max servo positions to 1000 and 2000. Also, make sure you understand my server linearization feature.  This not only eliminates binding in four servo swashplates, but also provides a linear relationship with a collective angle.  You’ll have to convince me that what you’re doing is better than what’s there already.

thanks again for your efforts in the Motors heli’s library. I appreciate the help.

Gone4Dirt, 2023-05-05:
@bnsgeyer no problem, happy to be able to help out :-)

Regarding limit cyclic angle:  Understood on your point.  To add clarification on where this is comming from:
We were setting up a coax heli (dual_mode == Intermeshing) which does use cyclic for both roll and pitch control.  This useage of only limiting pitch output:
https://github.com/ArduPilot/ardupilot/blob/a43680e88246bd2fa4b3f57b84318fccc7b2ae79/libraries/AP_Motors/AP_MotorsHeli_Dual.cpp#L519-L528
Gives the issue i describe above.  We now do not have the ability to set a cyclic range in roll and therefore must set the pitch limited cyclic range to match (if we want symetrical control).

Regarding servo min max concern.  Yes I totally agree, we need to understand why things were previously done in the past before changing them.  I propose we have a call to chat this through.

bnsgeyer, 2023-05-05:
@Gone4Dirt good catch. I was thinking strictly of tandem. The library will need to treat pitch and roll the same regardless of dual heli type.