ArduPilot/ardupilot #23712: Unable to compile

David-OConnor, 2023-05-07:
## Bug report

**Issue details**

Ardupilot compilation fails with the following error, after running `python3 waf copter`
```c
Generating C implementation in directory /mnt/c/Users/david/Desktop/ardupilot/build/MatekH743/libraries/GCS_MAVLink/include/mavlink/v2.0/cubepilot
Copying fixed headers for protocol 2.0 to /mnt/c/Users/david/Desktop/ardupilot/build/MatekH743/libraries/GCS_MAVLink/include/mavlink/v2.0
Waf: Leaving directory `/mnt/c/Users/david/Desktop/ardupilot/build/MatekH743'
Build failed
 -> task in 'dronecan' failed (exit status 2):
        {task 140616402921600: dronecangen  -> }
 (run with -v to display more information)
david@Data-Desktop:/mnt/c/Users/david/Desktop/ardupilot$
```

Steps to reproduce:
Using Ubuntu 22.04 (WSL in this case):

  - Clone the repo; cd into it
  - `python3 waf configure --board MatekH743 ` x 2 (It will tell you you need to run the command again the first time*)
  - `python3 waf copter`
  
The Matek board is an example, but is not required to induce the error.

* `configure` must be run twice due to this error the first time:
```c
Missing waf submodule. Trying to get it
Submodule 'modules/waf' (https://github.com/ArduPilot/waf.git) registered for path 'modules/waf'
Cloning into '/mnt/c/Users/david/Desktop/ardupilot/modules/waf'...
Submodule path 'modules/waf': checked out '1b1625b8e7da6e1307d73335cb995fa8813d5950'
Submodules OK, try running again
```

**Version**
Github, CAO 2023-05-07

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

andyp1per, 2023-05-07:
You are missing submodule setup
```
git clone https://github.com/ArduPilot/ardupilot.git
cd ardupilot
git submodule init
git submodule update --recursive
./waf configure --board MatekH743
./waf clean
./waf copter
```

closed 2023-05-07 by andyp1per

David-OConnor, 2023-05-07:
Thank you. This now results in a long red error of this form:

```c
make: *** No rule to make target 'CrashCatcher_armv7m.S', needed by 'modules/ChibiOS/obj/CrashCatcher_armv7m.o'.  Stop.
make: *** No rule to make target 'CrashCatcher_armv7m.S', needed by 'modules/ChibiOS/obj/CrashCatcher_armv7m.o'.  Stop.
make: *** Waiting for unfinished jobs....

Waf: Leaving directory `/mnt/c/Users/david/Desktop/ardupilot/build/MatekH743'
Build failed
 -> task in 'ChibiOS_lib' failed (exit status 2):
        {task 140349592364848: ChibiOS_lib hwdef.h,hw.dat,include_dirs,board.c,board.h,bouncebuffer.c,bouncebuffer.h,chconf.h,crashdump.c,ffconf.h,flash.c,flash.h,halconf.h,hrt.c,hrt.h,malloc.c,mcuconf.h,poll.h,ppm.h,spi_hook.h,stdio.h,stm32_util.c,stm32_util.h,stm32f1_mcuconf.h,stm32f3_mcuconf.h,stm32f47_mcuconf.h,stm32g4_mcuconf.h,stm32h7_mcuconf.h,stm32h7_type2_mcuconf.h,stm32l4+_mcuconf.h,stm32l4_mcuconf.h,stubs.c,usbcfg.c,usbcfg.h,usbcfg_common.c,usbcfg_dualcdc.c,watchdog.c,watchdog.h,chibios_board.mk,chibios_common.mk,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,buzzer.c,buzzer.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,b
 ```

I'm happy to post a new issue since it's different, and you solved the original, or work it here.

andyp1per, 2023-05-07:
You are missing the crash catcher submodule. Try running ./Tools/gittools/submodule-sync.sh

David-OConnor, 2023-05-07:
It works now; thank you!

Final recipe:

  - git clone https://github.com/ardupilot/ardupilot
  - cd ardupilot
  - git submodule init
  - git submodule update --recursive
  - ./Tools/gittools/submodule-sync.sh
  - python3 waf configure --board AnyleafH7
  - python3 waf clean (Optional?)
  - python3 waf copter

alccrl, 2023-08-07:
Hi,

I got same problem.

>>But I did not execute : <<
git submodule init
git submodule update --recursive

I attach main steps for sitl build.

See history summary here (ubuntu 23.04):
  -- history --
  
  (...)
  112  ./install-prereqs-ubuntu.sh -y
  (...)
  114  sudo apt-get install gitk git-gui
  115  git clone https://github.com/ArduPilot/ardupilot.git
  (...)
  120  cd ardupilot/
  121  ./waf list_boards
  122  ./waf configure --board=speedybeef4v3
  123  ./waf copter
  (...)
  147  ./waf distclean
  (...)
  149  ./waf configure --board=speedybeef4v3
  (...)
  153  ./waf copter
  (...)
  155  ./waf distclean 
  (...)
  167  ./waf configure --board=sitl
  168  ./waf copter
       --- errors
  (...)
  172  ./Tools/gittools/submodule-sync.sh
  173  ./waf -v copter
       --- build sitl ok

So, please review necessary steps and update build instructions on site, when possibble.

Thanks,

Alfredo

alccrl, 2023-08-07:
Important: 
I got failed build today (bug is already declared closed)

Alfredo

aogrcs, 2024-02-24:
May this link would work
https://discuss.ardupilot.org/t/dronecangen-returned-1-error/109920/6