ArduPilot/ardupilot #23714: Drone movement when changing Loiter from AUTO mode to GCS with RC off

Jee88, 2023-05-08:
## Bug report
**Issue details**

Hello.

I'm asking because it seems like there is a bug about changing to Loiter mode during automatic flight.

In this situation, when flying outside the signal range of the remote controller during long-distance flight, an obstacle is found in front, and in an emergency, if the mode is changed to Loiter through GCS, the drone moves slightly forward.

In order to test in the same situation, the FailSafe option was set to continue even if RC was cut off during automatic flight. The power of the controller was turned off during flight in autopilot mode. The drone continues to automatically fly towards the next WP.

At this time, if you change the mode to Loiter through GCS, the drone moves forward little by little. If you look at the log at this time, you can see that the GPS SPEED is 0.1~0.3m/s. I lowered the Loiter speed just to be safe, but a higher base speed makes it appear to move faster.

If you turn off the power of the RC controller during automatic flight and change the mode to Loiter while the MissionPlanner Joystick is connected, this phenomenon does not occur.

[image]
**Version**
 ArduCopter 4.4-dev 

**Platform**
 [■] Copter

**Airframe type**
Hex

**Hardware type**
CUAVv5 plus

**Logs**
https://drive.google.com/file/d/1kh1WsT1kswWwuHs6wBLD8z2Jz594Abhp/view?usp=sharing

IamPete1, 2023-05-08:
Copter just keeps using the last received RC inputs. You can make it even more exciting by switching off the RC at full stick deflection. 

Ultimately we should either clear RC input like we do on plane, or not let the vehicle switch to loiter with no RC.

rmackay9, 2023-05-09:
@IamPete1,

We do actually handle the case of no RC input in loiter mode as long as it's been detected as an RC failsafe.  You'll see that both in ModeLoiter::init(), ModeLoiter::run() and Mode::get_pilot_desired_lean_angles() we handle no RC input and set the desired lean angles to zero.

So I don't immediately see how the issue raised is happening, it shouldn't as far as I can see.  The only reason I can imagine it happening is if the RC failsafe has not triggered or we have a bug somewhere.

Jee88, 2023-05-09:
At the time of flight, my FS_OPTIONS was 31.
Even if it is RC failsafe in AUTO mode, the situation of Continue is compatible with the option.
[image]
The problem is that if you change to Loiter through GCS at this time, it may have to be changed to RTL mode because it is already RC off, but in this situation, RTL does not occur and hovering occurs.

I don't think this situation is too much of a problem.
This is because you may want to stop with Loiter before determining RTL directly through GCS when there is a risk of hitting an object ahead through video due to incorrect altitude setting during automatic flight.

However, in this case, I think it is correct that the drone moves in the situation where it is changed to Loiter as above.
(Of course, there is no problem if you connect it with a joystick when flying long distances)

Please let us know if you need additional flight status or conditions at the time.

peterbarker, 2025-08-12:
We've fixed this one; loiter mode only uses the RC inputs if they are valid.

https://github.com/ArduPilot/ardupilot/pull/25822 and others.