ArduPilot/ardupilot #23738: AP_Logger: stuck thread with WeBots SITL on Ubuntu 20.04

ZmeiGorynych, 2023-05-10:
## Bug report

**Issue details**
I run Copter-4.4 branch on Ubuntu 20.04
When I try to follow the [instructions for running SITL with WeBots](https://ardupilot.org/dev/docs/sitl-with-webots-python.html), everything starts up normally as far as I can tell, but get a critcal error `AP: AP_Logger: stuck thread ()` that I can't find the reason for. 

Same setup runs fine when simulating with FlightGear.

To reproduce: 
start webots, load iris_camera.wbt from the link above
```
$ /path/to/ardupilot/build/sitl/bin/arducopter -S -w --model webots-python --speedup 1 --slave 0 --defaults /path/to/ardupilot/Tools/autotest/default_params/copter.parm,/path/to/ardupilot/libraries/SITL/examples/Webots_Python/params/iris.parm --sim-address=127.0.0.1 -I0 &
$ mavproxy.py --master tcp:127.0.0.1:5760 --out udp:127.0.0.1:14550 --out udp:127.0.0.1:14551
```

**Version**
Copter-4.4, Ubuntu 20.04, Webots 2023a

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
webots-python

**Hardware type**
SITL with WeBots

**Logs**
Webots console looks healthy,
```
INFO: ardupilot_vehicle_controller: Starting controller: python3 -u ardupilot_vehicle_controller.py --motors "m1_motor, m2_motor, m3_motor, m4_motor" --camera camera --camera-port 5599
Camera stream started at 127.0.0.1:5599 (I0) (640x480 @ 10.00fps)
Listening for ardupilot SITL (I0) at 127.0.0.1:9002
Connected to ardupilot SITL (I0)
```
SITL seems to start fine otherwise
```
AP: AP_Logger: stuck thread ()
AP: GPS 1: detected as u-blox at 230400 baud
AP: GPS 1: detected as u-blox at 230400 baud
AP: AP_Logger: stuck thread ()
AP: GPS 1: detected as u-blox at 230400 baud
AP: AP_Logger: stuck thread ()
Flight battery 100 percent
AP: GPS 1: detected as u-blox at 230400 baud
AP: AP_Logger: stuck thread ()
AP: GPS 1: detected as u-blox at 230400 baud
AP: AP_Logger: stuck thread ()
AP: GPS 1: detected as u-blox at 230400 baud
AP: AP_Logger: stuck thread ()
Flight battery 100 percent
AP: GPS 1: detected as u-blox at 230400 baud
AP: AP_Logger: stuck thread ()
AP: GPS 1: detected as u-blox at 230400 baud
```

ZmeiGorynych, 2023-05-19:
@rmackay9 could this be related to https://github.com/ArduPilot/ardupilot/issues/17767?

sairaghavamunagala, 2024-04-29:
Try it by changing the baud rate.

AA-CambSensoriis, 2024-07-18:
Has anything happened with this? Facing the same issue

IanBurwell, 2024-08-31:
I just followed the provided commands on Ubuntu 22.04 and Copter-4.4.4 on my laptop without this issue arising. `AP_Logger: stuck thread` to my knowledge implies that some I/O is taking longer than 5 seconds ([here is the line that prints the error](https://github.com/ArduPilot/ardupilot/blob/1439aebf94ae388a82f28199aa7dd333c009b98c/libraries/AP_Logger/AP_Logger_File.cpp#L164), and [here is where I got the 5s number from](https://github.com/ArduPilot/ardupilot/blob/1439aebf94ae388a82f28199aa7dd333c009b98c/libraries/AP_Logger/AP_Logger_File.cpp#L1032)). Maybe this is due to a hardware limitation of the machines you all are running on?

Also, FWIW the `--speedup` parameter is not really respected by the webots implementation, as it was naively ignored and regardless of the speedup parameter you can run at max speed in webots with the `>>` button. If you are running at faster than real time in webots with that at the defaul value of 1, this does create even more logs without increasing the I/O timeout. That being said, one possible workaround (that may have other unforseen consequences) would be to set `--speedup` to a higher number as this increases the IO timeout proportionally.