ArduPilot/ardupilot #23778: Quadplane pitches down in auto mode after VTOL to Fixed-Wing transition done

jankrystianserafim, 2023-05-14:
## Bug report

_Please describe the problem_
It appears that the autopilot is giving the drone a command to maximally pitch down after completing the VTOL_TAKEOFF mission item and marking the Transition as "Done". This pitch down results in an altitude loss of even > 40m (until the Q_ASSIST_ALT is reached). This issue is almost definitely not a physical issue as it only happens in AUTO mode, during the transition from VTOL to fixed-wing, not in any other autonomous flight modes. For example, when the VTOL to fixed-wing transition is done from QLOITER to Guided mode or from QLOITER to RTL, there is no pitch down and no altitude loss at all. Please find the mission file, full parameter list and logs in the forum post below and at the Google Drive link. I have also described all the debugging and troubleshooting that I have done so far here:
https://discuss.ardupilot.org/t/quadplane-vtol-loses-altitude-during-transition-from-vtol-to-fixed-wing/100827

This is quite a dangerous issue, so any help and support here is much needed and appreciated. Thank you!

**Version**
4.1.1 dev
**Platform**
[ ] Plane
**Airframe type**
QUADPLANE
**Hardware type**
Cube Orange
**Logs**
https://drive.google.com/file/d/1EEdm0DMqo63MXu2ORtAnbaO-6lD7AA9F/view

Astik-2002, 2023-06-28:
The pitching moment could be due to vortex ring state build-up. I would like to work on this problem @jankrystianserafim. Could you share log files of your flights that were not done in Auto mode? It'll help me debug whether its a purely tuning issue or hardware issue.
Thanks

jankrystianserafim, 2023-06-30:
Thanks for getting back to us @Astik-2002 . I appreciate the help! Luckily for us, this problem has already been resolved. I will close this issue accordingly. You can read about the solution here: https://github.com/ArduPilot/ardupilot/issues/23866

closed 2023-06-30 by jankrystianserafim