ArduPilot/ardupilot #23815: AP_Camera/AP_Mount: improve Siyi driver's handling of take-picture failure feedback

rmackay9, 2023-05-18:
Our AP_Mount_Siyi driver processes failure feedback messages (aka FUNCTION_FEEDBACK_INFO messages) from the camera ([see here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Siyi.cpp#L426)) but all it does is display an error message to the user (this is also recorded in the onboard logs).  We should improve this handling especially for take-picture requests (e.g. DO_DIGICAM_CONTROL, IMAGE_START_CAPTURE commands) so that we do not incorrectly tell the GCS that a picture has been taken in cases where it actually fails (e.g. perhaps due to a missing or full SD card).

The enhancement could involve:

- [AP_Camera_Backend::check_feedback()](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Camera/AP_Camera_Backend.h#L104) would become virtual allowing other backends to override it.  AP_Camera_Mount would need to override it and call into a similarly named new method in AP_Mount/AP_Mount_Backend.  It would probably be best if this new function simply returned the number of pictures that had been taken or returned true if a picture had been taken since it was last called.
- AP_Mount_Siyi() would implement AP_Mount_Backend::check_feedback() and do something if it had received a FUNCTION_FEEDBACK_INFO recently.  Based on this the CAMERA_FEEDBACK message would be sent to the GCS (including the vehicle's location at the time the picture was taken, if available) and the CAM message would be recorded to the onboard logs.

Thanks to @jstroup for highlighting this issue.

DangerD1024, 2023-05-29:
Another BUG: If you set MNT1_TYPE = 8 but will not set any serial port to storm 32 gimbal you will get FC bootloop.
CSRF receiver should be also connected to achieve that.

rmackay9, 2023-05-29:
@DangerD1024,

Thanks for the report, I've added it to the Copter-4.4. issues list (see link above).