ArduPilot/ardupilot #23822: AP_DDS: Add service support

pedro-fuoco, 2023-05-18:
## Feature request

Refactor the AP_DDS client to allow services. To start out we could use `std_srvs/SetBool` to arm/disarm vehicles.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

SANTHOSH-MAMIDISETTI, 2023-06-09:
I would love to learn about this , and contribute to this , Can I be assigned this and briefed a bit on what has to be done , so that I might start working .   @pedro-fuoco   @Ryanf55

pedro-fuoco, 2023-06-09:
@SANTHOSH-MAMIDISETTI It might be a good Idea to talk to @arshPratap as he planned to solve this ticket in the future

SANTHOSH-MAMIDISETTI, 2023-06-09:
> @SANTHOSH-MAMIDISETTI It might be a good Idea to talk to @arshPratap as he planned to solve this ticket in the future

Thanks for the prompt response,@pedro-fuoco , if that's the case , can I be assigned to some other issue.

arshPratap, 2023-06-09:
@SANTHOSH-MAMIDISETTI  thanks for your interest in the project .. as @pedro-fuoco said above..I am currently working on the said project and have added the issue ticket [here](https://github.com/orgs/ArduPilot/projects/4/views/1?pane=issue&itemId=24321031) .. you can check the former link periodically to keep a track of the progress we make. Also you can take a look at this [repo](https://github.com/arshPratap/ROS2_DDS_Service_Demo) which is going to be a big testing ground for some of ROS-2 features we are planning to develop.Would definitely love to know your thoughts on the ROS-2 features that we will add in the future.
Thanks !

arshPratap, 2023-06-09:
> > @SANTHOSH-MAMIDISETTI It might be a good Idea to talk to @arshPratap as he planned to solve this ticket in the future
> 
> Thanks for the prompt response,@pedro-fuoco , if that's the case , can I be assigned to some other issue.

@SANTHOSH-MAMIDISETTI  you can find plenty of issues to contribute [here ](https://github.com/ArduPilot/ardupilot/labels/good%20first%20issue)

srmainwaring, 2023-09-05:
Closed by https://github.com/ArduPilot/ardupilot/pull/24145

closed 2023-09-05 by srmainwaring