ArduPilot/ardupilot #23842: Vulnerability - AP_GPS_GSOF latitude conversion subject to arithmetic exception

Ryanf55, 2023-05-21:
## Bug report

**Issue details**

While writing a simulator for the AP_GPS_GSOF driver, I forgot to swap the endianness in the latitude field for the position (GSOF2) packet. 

The following stack trace shows the issue. 
[image]

If the Trimble GPS sends incorrect data, it could cause a crash in the autopilot. The checksum does (mostly) protect against invalid data, so this would be protecting against errant values for the LLA field. If the GPS loses signal, it is not clear in the API whether it sends 0's, NaN, or something else, so the behavior is ill-defined. The API has actually removed this packet entirely, but the same lack of documentation is present for other fields. 

**Version**
e7b369473b67c8015153881f354bd97f65f4df9c

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
Ubuntu 22.04

rmackay9, 2023-05-21:
nice find.  by the way, the floating point exceptions will cause the simulator to crash but probably won't crash a real autopilot.

Ryanf55, 2023-05-22:
> nice find. by the way, the floating point exceptions will cause the simulator to crash but probably won't crash a real autopilot.

Thanks. I heard some devs fly in debug mode recently; this being the reason we are not allowed `assert` in the main loop. What is the desired behavior and protection for floating point exceptions, and are there any tests I should add to verify it that run in debug and/or release mode?

Ryanf55, 2023-05-23:
Can look at SIM_FLOAT_EXCEPT. 

To fix, add constrained float before reading msg.data value. Need an is_within_range for locations that returns a value. Reject the packet if it's true.

peterbarker, 2025-08-12:
> Can look at SIM_FLOAT_EXCEPT.
> 
> To fix, add constrained float before reading msg.data value. Need an is_within_range for locations that returns a value. Reject the packet if it's true.

Do you think we really should spend flash on those checks?  The GPS has to stuff up *or* the byte has to be corrupted in special ways.  And if it happens... we'll just rumble on and get things right for the most part anyway, squash invalid values elsewhere.