ArduPilot/ardupilot #23853: Very short Radio Failsafe

NikosSprocket, 2023-05-22:
Intro
Below is existing conversations from Discord with @Andy PIper#7618 about a peculiar issue TruthSeeker#6765 and me (NikosSprocket#1873) having lately with my X500 size Quadcopter and almost random "Radio Failsafe" messages that recover less than a second most of the time.

Summary
I have built a X500 size quadcopter Arducopter with ELRS Tx/Rx. I had the quad flying smoothly and performed many Stabilize, RTLs, SmartRTLs missions with excellent results. One day, due to user error I hit the Disarm switch during flight and the Quad fell from the sky on dirt. Damage was minimal, 1 leg support and 1 prop. I printed new parts and replaced arm holders, just to be on the safe side. Then tested all electronics GPS compass, IMUs operating perfectly and hovers or and static powered up MP readings where good except I start seen the short "Radio Failsafe" and recoveries.

Since then I have replaced the ELRS Rx without any change on this behavior. I also got more Failsafes with Arducopter version 4.3.6, 4.3.5 and V4.5.0-dev more frequent the Radio Failsafes than with 4.2.3.

The last test was done using 4.23 and sitting in the garage floor with a battery connected and had a dgps lock but did not have any Radio Failsafes looking at the MP through Sik telemetry. The next day I went to the flying field and within a few minutes of Loiter it went to Radio Failsafe and RTL. 

I am waiting for a new FC controller to eliminate any hardware issues.

Firmware: MatekH743-bdshot

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X

**Hardware type**
Matek H743-Slim V3 FC
Matek AP_PERIPH GNSS M10-L4-3100
Matek X Class 12S PDB
Radiomaster TX16S EdgeTX v2.8.3 - ELRS 3.2.1
Radiomaster Rx RP3 2400RX ELRS 3.2.1
T-Motor F35A 32bits 3-6S ESC

**Logs & Param**
https://drive.google.com/file/d/1gz7OrcenU8vpRfvXWnr59WQJA0NOG699/view?usp=sharing
https://drive.google.com/file/d/17LE09o2pxsSev9RvvZKdF8fnqkCpFu3S/view?usp=sharing
[v4-2-3_05-17-2023-param.zip](https://github.com/ArduPilot/ardupilot/files/11527623/v4-2-3_05-17-2023-param.zip)
[v4-3-6-NS-param.zip](https://github.com/ArduPilot/ardupilot/files/11527635/v4-3-6-NS-param.zip)

holsthei, 2023-05-28:
I've had the same issue, frequent radio failsafes (every couple of minutes just when having the drone sitting on the table and the transmitter within 1-2 meters).
Same flight controller (Matek H743-Slim V3), Radiomaster Zorro and Radiomaster RP3.
Running Ardupilot Copter 4.3.6 and ELRS 3.2.1 (EU LBT).

The only workaround I've found is setting ELRS packet rate to "100Hz Full". It fixes the issue (or at least makes the failsafes happen less often) - I haven't observed any in that mode so far.

Also tried a custom Ardupilot build to enable DMA on the FC's SERIAL7. I did not have any effect.

Unsure if it's an Ardupilot or ELRS issue, I'll try to get some logs if it still happens.

NikosSprocket, 2023-06-22:
New FC has arrived and installed. Same issues using F500. Changed to 250Hz and Telemetry Ratio 1:4, needed for yaapu smooth operation, and for the short time bench and short flight I have not seen an Radio FC.
When I have more flight time, I will let you know if I see any changes on this subject.
btw. Contacted Matek support and they point me to [https://github.com/ExpressLRS/ExpressLRS/issues/1773 ](url)

shiv-tyagi, 2023-06-28:
@andyp1per Just to report that I am also facing the same problem of frequent short radio failsafes and recoveries with Radiomaster tx16s MKII ELRS radio and RP1 receiver. I am using Cube Orange. The hack which @holsthei mentioned seems to work. I switched the packet rate to _100 Hz Full_ and the failsafe were very less frequent (only once in five minutes when I observed).

andyp1per, 2023-06-28:
@NikosSprocket you do not have the ELRS 420k options bit set. So you need:
```
RC_OPTIONS=8192
```
at a minimum, but obviously more bits if you want other things enabled.
@shiv-tyagi @holsthei do you have this bit set?

shiv-tyagi, 2023-06-29:
@andyp1per No, the bit was not set earlier. Thanks for letting me know. Tested this on bench after setting the mentioned bit. No falisafe reported in 10 minutes. Will post here if I see anything strange. 
Thanks again.

holsthei, 2023-06-29:
I have the bit set (RC_OPTIONS=8736).

andyp1per, 2023-06-29:
I wasn't able to provoke failsafes with this patch - https://github.com/ArduPilot/ardupilot/pull/24170 - I would be grateful if you could try it.

4.5-dev copter builds:

- MatekH743 - 
[arducopter.zip](https://github.com/ArduPilot/ardupilot/files/11910250/arducopter.zip)

- MatekH743-bdshot - 
[arducopter.zip](https://github.com/ArduPilot/ardupilot/files/11910228/arducopter.zip)

holsthei, 2023-07-01:
Thanks for the patch! I did a quick couple of minutes test with this 4.5-dev firmware and looks like it fixes the issue (or at least improves the situation) - I did not see any radio failsafes during the test in the Full 333Hz or F1000 modes I tried.

(With the 4.3.7 firmware the 333Hz Full mode caused a radio failsafe almost immediately)

andyp1per, 2023-07-01:
Thanks for testing! I will see whether this affects CRSF at all, but glad it improves things.

holsthei, 2023-07-01:
Based on the comments in the patch I might not be confident to run in F1000 mode if the max supported rate is 500 Hz even if it seemed to work in the short test... Not a problem for me since I was planning to use the 333Hz mode anyway, but maybe it should be mentioned in the documentation somewhere.

andyp1per, 2023-07-01:
F1000 is guaranteed to fail I would say - we would have to increase the rcin loop rate. It can be done, but is increased processor load.

timtuxworth, 2023-07-01:
> F1000 is guaranteed to fail I would say - we would have to increase the rcin loop rate. It can be done, but is increased processor load.

Could this be addressed by documentation? I'd be happy to increase the loop rate on my H743 planes to get this to work.

andyp1per, 2023-07-01:
> > F1000 is guaranteed to fail I would say - we would have to increase the rcin loop rate. It can be done, but is increased processor load.
> 
> Could this be addressed by documentation? I'd be happy to increase the loop rate on my H743 planes to get this to work.

I think, as we do for IMUs, increasing the rcin loop rate to 2Khz on H743 is a reasonable thing to do

holsthei, 2023-08-02:
Any chance to get the fix in 4.4 ?

peterbarker, 2026-11-15:
Closed by [PR #26183](https://github.com/ArduPilot/ardupilot/pull/26183) and [PR #24283](https://github.com/ArduPilot/ardupilot/pull/24283). The CRSF parser now resyncs on the next frame marker instead of waiting for a timeout (b071d76565), so a delayed or corrupted ELRS frame costs one frame rather than triggering a failsafe. CRSFv3 baud rates are also rescanned after a soft reboot (15047ddfc7). Please reopen if short failsafes still happen on 4.5 or later with RC_OPTIONS bit 13 set.