ArduPilot/ardupilot #23859: Why do new objects not need to be deleted?

xianglunkai, 2023-05-23:
for example:  AP_RangeFinder
 AP_RangeFinder_Backend *drivers[RANGEFINDER_MAX_INSTANCES];
The object pointed to by this pointer array is obtained through new operation, why not release it in the destructor function？
Also, I have checked many places and found that this is the case. Is there any mechanism that I overlooked？
Thank you very much!

rmackay9, 2023-05-23:
Hi @xianglunkai,

No, there's no need to delete objects because the destructor should never be called.  As a general use in AP we try not to create any new objects in flight due to the risk of the new failing and leading to memory corruption or attempts to access uninitialised memory.  This is why you'll see that most objects are created soon after startup in one of the many init() methods.

closed 2023-05-23 by rmackay9

xianglunkai, 2023-05-23:
@rmackay9 
Thank you very much!