ArduPilot/ardupilot #23874: MAV_CMD_CONDITION_YAW inconsistencies - for relative turns.

AndKe, 2023-05-24:
## Bug report

**Issue details**
This is my test-mission used with SITL:
[yawtests.txt](https://github.com/mavlink/mavlink/files/11405655/yawtests.txt)

These are my findings:
https://docs.google.com/spreadsheets/d/1FJebVrADxDw2-KrqyvU_h31yHybQQziYpFmSMtuBGWA/edit?usp=sharing

I indicated earlier that I thought Absolute heading changes with "shortest way" worked.
..And it does..

The same cannot be said about relative heading changes that are not even trying to force a particular direction,
but I discovered a very strange behavior on mission item 22, commanding relative -90° from an actual heading of 90° ends somehow up at 180° heading.

**The objective**
The current heading changes with relative headings are actually working as in my PR https://github.com/mavlink/mavlink/pull/1962#issuecomment-1560451836
which is useful, and intuitive.

The only remaining issue is that there is a bug in how relative headings are changed (with negative number of degrees - which should cause a relative heading to be achieved by turning counterclockwise)

**Version**
AC (master)

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X4

**Hardware type**
SITL

**Logs**
see https://github.com/mavlink/mavlink/pull/1962#issuecomment-1536150878 for more information.

hamishwillee, 2023-05-25:
I can verify from testing that if you're facing 90 degrees (east) and command -90° as a relative yaw you don't end up at zero, but instead at 180.

The problematic code is here: https://github.com/ardupilot/ardupilot/blob/master/ArduCopter/autoyaw.cpp#L116 - as explained in [the comment](https://github.com/mavlink/mavlink/pull/1962#issuecomment-1560451836) there's a multiplication of two negatives turning the direction into a +90 degree relative turn.

The linked MAVLink PR is to allow a value of 0 on the direction field to make the vehicle head to the relative yaw in the direction with the smallest angle, rather than specifying the direction. Can that be fixed at the same time?

@auturgy this is the bug we were talking about in the mav call.

rmackay9, 2023-11-01:
Hi, sorry for missing this.  I've added this to the [4.4 issues list](https://github.com/ArduPilot/ardupilot/issues/23192).

rmackay9, 2023-11-10:
I've looked into this a bit and commented on the MAVLink PR. https://github.com/mavlink/mavlink/pull/1962

Here are my comments again.

I think the confusion on the AP side is because of the target angle (param1) is apparently always suppose to be between 0 and 360.  This has led to an inconsistent interpretation of the direction field.  In AP, for a relative, CCW rotation (e.g. rotate left) you don't input a negative angle, instead you put in the angle (e.g. 90) and direction of -1 (e.g. CCW).

`<param index="1">target angle: [0-360], 0 is north</param>`

We could change it but I think AP does allow a user to get what they want.  For example if they wanted the vehicle to rotate 270 degrees CCW they would input angle=270, direction=-1.

One bug in AP is that when using mission commands it does not reject negative angles.  It does reject negative angles when the commands are received as immediate commands (e.g. within MAVLink COMMAND_LONG or COMMAND_INT messages)

So I guess we have two choices:

1. fix the target angle description to accept angles in the -180 to +360 range, change the AP implementation
2. leave it as-is and other stacks adopt the AP method, AP adds the negative target angle check for mission commands

FYI @lthall

hamishwillee, 2023-11-15:
I've counter commented here https://github.com/mavlink/mavlink/pull/1962#issuecomment-1811873068 - I think we should do "2" but needs to be tidied up in the mavlink/mavlink repo and then pushed down so that it matches with ardupilot/mavlink. 

Why? Because I think the upstream used to match this definition and was changed accidentally, there is no one else implementing this (AFAIK), and it makes sense. I am hoping @rmackay9  will create the upstream PR, but if not let me know and I'll take a shot at it.

tzi4, 2026-08-20:
I was reading the code related to auto yaw when this caught my eye, and I wanted to ask about it. I am trying to understand the historical intent here. So in mavlink/mavlink#1962, it was noted that Arducopter already used the shortest direction for absolute direction=0, but treated relative direction=0 as CW. An Arducopter implementation change was initially requested before the Mavlink definition was eventually merged. The replacement, which is mavlink/mavlink#2058, was later merged with param1=0-360 and a general param3=0 definition for the shortest direction, under the assumption that this matched Ardupilot. However as far as I understood, the current Copter code and the Copter wiki still retain the legacy relative 0=CW behaviour.
On current master (2cd33afd), relative 270/0 produces about +271° and relative 360/0 produces about +362°. I was thinking of using wrap_PI() for the relative zero direction path, which produces about -90° and 0°.  With a small adjustment in guided to use explicit +1/-1, guided signed offsets remain unchanged while absolute yaw is unaffected.
But after reading all the related posts I am kind of confused, so could someone please clarify which way was intended going forward? Was it decided that preserving the documented Copter legacy behaviour (relative 0=CW) and qualifying the Mavlink/QGC definitions accordingly is the right way, or should making relative direction=0 use the shortest direction, as specified by the current common.xml, be the way to go?
I have a small tested patch for the second way and would be happy to open a separate PR if that is preferred. If not, sorry for taking your time I am just trying to understand.