ArduPilot/ardupilot #23878: SITL with Realflight is not working properly

ParasInternKhushPatil, 2023-05-24:
## Bug report

**Issue details**

I have launched the sim_vehicle.py with flightaxis command 
It shows that flightaxis controller device is restored & activated but in mav console we are getting link 1 down 
Sir we have properly configured the networking
But the this issue still exists
the Networking is properly configured as mentioned in The documentation
**Version** 

Current

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[*] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QuadCopter X & AltiTransition FPV Gimbal

**Hardware type**
SITL 

**Logs**
sim_vehicle.py -v ArduCopter -f flightaxis:192.168.***.** --map --console
SIM_VEHICLE: Start
WARNING: no config for frame (flightaxis:192.168.***.**)
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/kp/ARDupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/kp/ARDupilot 
Setting out to                           : /home/kp/ARDupilot/build 
Autoconfiguration                        : enabled 
Checking for program 'python'            : /usr/bin/python 
Checking for python version >= 3.6.9     : 3.8.10 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
CXX Compiler                             : g++ 9.4.0 
Checking for need to link with librt     : not necessary 
Checking for feenableexcept              : yes 
Enabled OpenDroneID                      : no 
Enabled firmware ID checking             : no 
GPS Debug Logging                        : no 
Enabled custom controller                : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Configured VSCode Intellisense:                : no 
DC_DSDL compiler                               : /home/kp/ARDupilot/modules/DroneCAN/dronecan_dsdlc/dronecan_dsdlc.py 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Debug build                                    : disabled 
Coverage build                                 : disabled 
Force 32-bit build                             : disabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (1.451s)
{'model': 'flightaxis:192.168.137.27', 'sitl-port': False, 'waf_target': 'bin/arducopter'}
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/kp/ARDupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter"
Waf: Entering directory `/home/kp/ARDupilot/build/sitl'
Embedding file locations.txt:Tools/autotest/locations.txt
Embedding file models/Callisto.json:Tools/autotest/models/Callisto.json
Embedding file models/plane-3d.parm:Tools/autotest/models/plane-3d.parm
Embedding file models/plane.parm:Tools/autotest/models/plane.parm
Embedding file models/xplane_heli.json:Tools/autotest/models/xplane_heli.json
Embedding file models/xplane_plane.json:Tools/autotest/models/xplane_plane.json
Waf: Leaving directory `/home/kp/ARDupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/kp/ARDupilot/build/sitl
Target          Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)  External Flash Used (B)
----------------------------------------------------------------------------------------------------------
bin/arducopter   3832418    166021   212008               3998439  Not Applicable  Not Applicable         

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (2.828s)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/home/kp/ARDupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/home/kp/ARDupilot/build/sitl/bin/arducopter" "-S" "--model" "flightaxis:192.168.137.27" "--speedup" "1" "--slave" "0" "--sim-address=127.0.0.1" "-I0"
RiTW: Starting ArduCopter : /home/kp/ARDupilot/build/sitl/bin/arducopter -S --model flightaxis:192.168.***.** --speedup 1 --slave 0 --sim-address=127.0.0.1 -I0
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--master" "tcp:127.0.0.1:5760" "--map" "--console"
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Loaded module map
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760

jkronk, 2023-06-28:
I have got the same issue, but in XPlane. I think this is a docker networking issue. I have tried doing the --network=hosts but that only works in linux. 

My issue is the heartbeat as well, my mission planner to SITL connection can be made, but not to XPlane. I think something similar is happening to you.

peterbarker, 2025-08-12:
So RalFlight is well tested and working.

Did you resolve your problems here?