ArduPilot/ardupilot #23893: Boat: Support boat simulation at a certain flow rate

xianglunkai, 2023-05-25:
In my more than seven years of unmanned ship projects, a large portion of them were located on the Yangtze and Yellow Rivers, where the water flow speed was high. I believe that AP needs to support simulation in this environment, which has great significance for debugging ship loiter mode and auto mode.
     It is worth mentioning that I first replaced the PID controller on the AP with an active disturbance rejection controller in these scenes, ensuring accurate ship control even under large disturbances.

IamPete1, 2023-05-25:
The sailboat simulator can do tide with `SIM_TIDE_DIR` and `SIM__TIDE_SPEED`. You can also simulate waves with the `SIM_WAVE` parameters. They only kick in once the vehicle is armed. 

It should be quite easy to bring them into the normal boat simulator as well.

xianglunkai, 2023-05-26:
@IamPete1 
Thank you!  I will try it!

closed 2023-05-26 by xianglunkai