ArduPilot/ardupilot #23897: AP_Scripting: reduce need to run mavgen to support sending/receiving mavlink messages

rmackay9, 2023-05-26:
PR https://github.com/ArduPilot/ardupilot/pull/13660 adds support for sending arbitrary mavlink messages but it requires using mavgen to create a lua library which includes the message definitions and the mavlink object's decode_header, decode and encode methods.

Requiring the generation of this extra library and then struggling with the require() is more than many script developers will be comfortable doing so I think we should find a way to avoid this.  One option is to simply create the files in github and update them periodically as the message definitions change.

We should also add a simple example that shows how to send and receive as-is with no extra steps.  I don't think the current examples work as-is.

WickedShell, 2023-05-26:
Because of memory reasons you actually want to compile the minimum set of messages you need, or reduce them after generation manually. The most useful thing for the user would probably be a page we host that generates the exact sub set of messages a user cares about and offers it to them as a download.

-------- Original Message --------

IamPete1, 2023-05-26:
I was thinking we just put each message in a separate file the you can put them all on the SD card and just require the ones you need. We could host such a folder somewhere.

WickedShell, 2023-05-26:
> I was thinking we just put each message in a separate file the you can put them all on the SD card and just require the ones you need. We could host such a folder somewhere.

You could if you add a meta helper that pushes the tables together nicely I guess. Unless something changed and we don't require them to be all in the same table together anymore? It's not the biggest deal to combine the tables, but we do need the indexes to match the id's which the generated code had been nicely handling for us.

There's also an incredibly long term problem with that path that we can ignore, but there's a limit to the number of files we can stick in a folder on the SD card, and on a FAT32 card it's less then we can technically have. (In practice it's so far away though I think we can ignore this, by the time we have 65,535 some messages I think we will probably have had to do something about the firmware impacts :) )

peterbarker, 2025-08-12:
Turns out from a practical standpoint that in-lining the messages by hand is the most practicable solution at this point; heard enough for people to install scripts, let alone specific dependencies.  Certainly the last time I went to do mavlink in a script inlining was recommended.