ArduPilot/ardupilot #23901: Plane/Blimp: populate AP_Vehicle::is_landing() and AP_Vehicle::is_taking_off()

magicrub, 2023-05-26:
PLANE needs to populate AP_Vehicle::is_landing() and AP_Vehicle::is_taking_off() which was recently added by @tatsuy in PR
 https://github.com/ArduPilot/ardupilot/pull/23827 because right now it's always false.
 
 Here's some useful logic that someone could implement:
 
Plane.is_landing() == true when
  - in AUTO and current mission item is NAV_LAND or some VTOL ones
  - VTOL library says we're landing

Plane.is_taking_off() == true when
  - in mode AUTO and current mission item is NAV_TAKEOFF or some VTOL ones
  - in mode TAKEOFF
  - VTOL library says we're taking off
  
  
 Check Copter's logic to try and match it's behavior too.. like maybe being ARMED is required to make it true? Being in airborne aka plane.is_flying()?

@MichelleRos Does Blimp have logic that could populate these too? If not lets remove the label.

MichelleRos, 2023-05-27:
At the moment landing is still an issue for blimp as I have no idea how we would detect it - or even define it. Even touching the ground is not necessarily "landed" because the next breeze or air current can move it away. The only real "landing" a blimp does is when the operator attaches it to something. 

For is_landing I guess we could set it during land mode (although that wouldn't capture landing in eg manual or loiter), but taking off would have the same problem I mentioned above.

BabaYaga1221, 2023-05-28:
Hello, I am interested to contribute in this project. Can you assist me with this work and help me to understand the files which I need to work on? Thanks

flatherz, 2023-06-02:
I am interested in working on this issue. Could I be assigned to it?

SANTHOSH-MAMIDISETTI, 2023-06-09:
Hello, I am interested to contribute .Could I be assigned to it? and briefed @magicrub @MichelleRos

magicrub, 2023-06-14:
@flatherz  Thanks for https://github.com/ArduPilot/ardupilot/pull/24021 PR!

SANTHOSH-MAMIDISETTI, 2023-06-15:
> @flatherz Thanks for #24021 PR!

So, is this issue solved ?@magicrub

SANTHOSH-MAMIDISETTI, 2023-06-15:
> > @flatherz Thanks for #24021 PR!
> 
> So, is this issue solved ?@magicrub

@flatherz

magicrub, 2023-06-15:
@SANTHOSH-MAMIDISETTI well he has submitted a code change. It needs a couple minor changes but otherwise it's done-ish.

SANTHOSH-MAMIDISETTI, 2023-06-15:
> Until that PR is merged, the issue is still unresolved
Oh okay @magicrub  , If that is the case , can I go and look into issues for some new issue to work on ?

magicrub, 2023-06-15:
@SANTHOSH-MAMIDISETTI you can do whatever you want :)

closed 2023-06-26 by magicrub