ArduPilot/ardupilot #23902: Modified files in pymavlink submodule on fresh clone of master (crlf line endings)

casrya, 2023-05-27:
## Bug report

**Issue details**

I noticed this today and it looks like someone might have accidentally committed .py files under pymavlink/ submodule with crlf line endings?

To confirm, do a fresh clone with git clone https://github.com/ArduPilot/ardupilot.git --recurse-submodules
(I tested at commit edf9fbdb60bc5730ed928507bd8148b9b0b6dfa2)

Change to pymavlink/ submodule folder and do a git status and you should see modified .py files

cat -A on one of the files and you should see it has ^M (CR) which appears to be the only change

P.S I noticed this when I did a merge on my own code so did the fresh clone to confirm. Hopefully someone can confirm either way since if it is just me I will have to work out what happened ;)

**Version**
master at commit edf9fbdb60bc5730ed928507bd8148b9b0b6dfa2

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

magicrub, 2023-05-27:
Yeah, that seems to concur with what I'm seeing. We need to git blame it!

casrya, 2023-05-27:
@magicrub thanks for the quick update, Cheers

peterbarker, 2025-08-12:
I think this was sorted way back when.  Please re-open if I'm wrong.

closed 2025-08-12 by peterbarker