ArduPilot/ardupilot #23913: waf distclean not removing build/ folder after commit 22c1470acac8bc112e77ffcdfb868314a3d32be7?

casrya, 2023-05-29:
## Bug report

**Issue details**

waf distclean not removing build/ folder after commit 22c1470acac8bc112e77ffcdfb868314a3d32be7?

I discovered this today when I was trying to build latest merged master and had compile errors in libraries/AP_DroneCAN and related files.

I tracked it down to being because the old generated libcanard files under modules/DroneCAN/libcanard were not being cleaned. After a bit more investigation I realized that it was due to commit 22c1470acac8bc112e77ffcdfb868314a3d32be7 such that the original 'rm -rf build' does not seem to be executed anymore on ‘./waf distclean’, only the submodule sync is performed.

I am not experienced with waf so don’t have a fix other than manually deleting build/ when a distclean is required but reporting FYI.

P.S I am building on wsl2 Ubuntu 20.04.5 LTS if this is relevant?

Cheers.

**Version**
master at commit e415b5fda0c0ecec546f0395c6cbb6d69802a5cf

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

rmackay9, 2023-05-29:
FYI @bugobliterator @magicrub

tridge, 2023-05-30:
@bugobliterator can you look?

magicrub, 2023-05-30:
Shall we just revert for now?

rmackay9, 2023-05-31:
Fixed in master so closing, thanks for the report!

closed 2023-05-31 by rmackay9