ArduPilot/ardupilot #23917: git clone git://github.com/ArduPilot/ardupilot.git - Unknown error

DangerD1024, 2023-05-29:
`git clone git://github.com/ArduPilot/ardupilot.git
Cloning into 'ardupilot'...
fatal: unable to connect to github.com:
github.com[0: 140.82.121.4]: errno=Unknown error`
Can't download it locally, works fine on other repos on github...
What i'm doing wrong?))

casrya, 2023-05-29:
try **https**, this is what I do on wsl2 (you may or may not want the recurse-submodules but probably do for a fresh clone)

`git clone https://github.com/ArduPilot/ardupilot.git --recurse-submodules `

rmackay9, 2023-05-29:
I think this was probably a momentary issue so I'll close this if that's OK.  I'll monitor though in case there are more replies and we can re-open if necessary.

closed 2023-05-29 by rmackay9

casrya, 2023-05-30:
@rmackay9, understood, I just replied to the original poster because I ran into this a while back.

FYI Out of interest I just tested again with **git://** (since I always use **https://**) and this is what I got after a few mins to timeout (but https worked straight away). 

```
fatal: unable to connect to github.com:
github.com[0: 20.248.137.48]: errno=Connection timed out
```

I wouldn’t be surprised if it is a wsl2 network/firewall issue though.

That said the docs at https://ardupilot.org/dev/docs/git-clone.html do already list the git URL with https so the docs already have it covered :)

Cheers.

magicrub, 2023-05-30:
The git:// protocol has been disabled by GitHub because it's not secure.