ArduPilot/ardupilot #23921: Compilation warnings for AP_DroneCAN

Ryanf55, 2023-05-29:
## Bug report

**Issue details**

Compilation warnings/errors have been introduced in AP_DroneCAN from a narrowing conversion.

**Version**
Master hash `64292958694d532b6bb92d6c8ea7c5210fc5c5a1`

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
```
$ rm -r build
$ colcon build
../../libraries/AP_DroneCAN/AP_Canard_iface.cpp: In member function ‘virtual bool CanardInterface::broadcast(const Canard::Transfer&)’:
../../libraries/AP_DroneCAN/AP_Canard_iface.cpp:64:39: error: narrowing conversion of ‘(uint32_t)bcast_transfer.Canard::Transfer::payload_len’ from ‘uint32_t’ {aka ‘unsigned int’} to ‘uint16_t’ {aka ‘short unsigned int’} [-Werror=narrowing]
   64 |         .payload_len = bcast_transfer.payload_len,
      |                        ~~~~~~~~~~~~~~~^~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors
```

Ryanf55, 2023-05-29:
@bugobliterator Would you be able to issue a fix for this? The blame shows this was introduced in commit: https://github.com/ArduPilot/ardupilot/commit/0c728972ff0690737e2f455b5808667915e8f597

peterbarker, 2025-08-12:
Looks like tridge fixed this in e1dd452b66257208f50c2395a58a8728f43626bb

Ta!