ArduPilot/ardupilot #23927: Set Servo options triggered by failsafe

alexanderever, 2023-05-30:
## Feature request

For UAS which perform a function controlled by Set_servo outputs, such as crop spraying, it would be desirable for the UAS to stop doing the function if a failsafe mode is triggered. It is not clear if the Sprayer functionality includes this but it may be useful for other  applications such as vectored thrust VTOLs etc

Currently the consequence of RC signal loss while spraying would be the UAS performing a RTL while still spraying. This would be VERY BAD as the spray would be released at a high altitude and subsequently drift to unintended locations. 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

A parameter to set servo outputs should failsafe be triggered Eg SERVOXX_FS_VALUE = 1000 

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Lua scripting?

checked the Sprayer functionality in ardupilot, it is unclear if this issue is addressed. 

(Also documentation on "SPRAY_SPINNER Spinner rotation speed: Spinner's rotation speed in PWM (a higher rate will disperse the spray over a wider area horizontally)" is inaccurate and could cause serious issues, increased RPM actually creates smaller droplets.  see https://en.wikipedia.org/wiki/Rotary_atomizers)

**Platform**
[ x ] All
[x  ] AntennaTracker
[ x ] Copter
[  x] Plane
[ x ] Rover
[  x] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

rmackay9, 2023-05-30:
Thanks for the feature request.  Turning off the sprayer during any kind of failsafe makes a lot of sense.

alexanderever, 2023-05-31:
Hi Randy,

thanks for considering the feature!

 I work for an agricultural company with a long history of spraying in ground based and aerial settings. We also contribute to international discussions on drones as spraying systems. There are a number of features we can think of that we feel would improve Arducopter equipped aircraft for this role.

i'd be very interested in helping the development of Arducopter.

Please feel free to contact me at ***@***.***

Thanks,

Alexander

peterbarker, 2025-08-12:
@alexanderever a script has been offered up to accomplish what you wanted here.  Would you be happy with such a solution?

rmackay9, 2025-08-13:
I agree with @alexanderever that we should just add it as a SPRAYER_OPTIONS parameter because this request has come up so many times.

Priyanshu-Shah, 2025-12-23:
I have added the battery and RTL failsafe as well, it just turns off the servos if battery levels reach critical or when any type of RTL action is performed for any failsafe. 

For clarity, not all failsafes are critical, so currently, sprayer turns off if:
- RC/GCS failsafe
- Battery Critical failsafe
- Any type of failsafe but the action is RTL based or landing based. 

@rmackay9 @peterbarker Requesting feedback on the height based cap. What kind of an implementation is suggested?

shashwatvoa, 2026-02-15:
I just made a pull request adding this fearure can someone review it and guide me as I am new here