ArduPilot/ardupilot #23939: Wrong version of python launched in `sim_vehicule.py` after installation with `environment_install` tool on Windows

julianmaster, 2023-05-31:
## Bug report

**Issue details**

After running the `environment_install` tool for Windows `install-prereqs-windows.ps1`, I follow the [Copter SITL/MAVProxy Tutorial](https://ardupilot.org/dev/docs/copter-sitl-mavproxy-tutorial.html) and running this command give me an error : 
```
$ ../Tools/autotest/sim_vehicle.py --map --console
Traceback (most recent call last):
  File "/cygdrive/c/Users/jmaitr03/Documents/Github/ardupilot/ArduCopter/../Tools/autotest/sim_vehicle.py", line 31, in <module>
    from pysim import util
  File "/cygdrive/c/Users/jmaitr03/Documents/Github/ardupilot/Tools/autotest/pysim/util.py", line 19, in <module>
    import pexpect
ModuleNotFoundError: No module named 'pexpect'
```

I found that in the sim_vehicule.py script, the shebang is wrong because it says :

```
#!/usr/bin/env python3
```

In the `install-prereqs-windows.ps1` script, a symbolic link are made to python3.7 but with the name `python` and not `python3` :

```
Start-Process -wait -FilePath "C:\cygwin64\bin\bash" -ArgumentList "--login -i -c 'ln -sf /usr/bin/python3.7 /usr/bin/python'"
Start-Process -wait -FilePath "C:\cygwin64\bin\bash" -ArgumentList "--login -i -c 'ln -sf /usr/bin/pip3.7 /usr/bin/pip'"
```

This misconfiguration causes the error shown above.

**Version**
Ardupilot-git

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Windows

xli-li, 2024-08-05:
You will, please do you have a solution ?

xli-li, 2024-08-05:
Hello, do you have a solution?

lxyruanjian, 2024-09-04:
@xli-li I also encountered this issue and solved it using the following script:
`
Start-Process -wait -FilePath "C:\cygwin64\bin\bash" -ArgumentList "--login -i -c 'ln -sf /usr/bin/python3.9 /etc/alternatives/python3'"
`