ArduPilot/ardupilot #23973: Pose jump after source switching OPT_FLOW->EXT_NAV(Tertiary->Secondary) after EXT_NAV reset

Pallav1299, 2023-06-05:
## Bug report

**Issue details**

While trying to handle EXT_NAV(VIO) source resets during mission execution by using EKF Source Switching, I notice an abrupt pose jump while switching from OPT_FLOW->EXT_NAV  after EXT_NAV source reset in LOITER and GUIDED modes.
**Source list:**
* Primary: GPS
* Secondary: EXTNAV(VIO)
* Tertiary: OPT_FLOW
[image]

**Workflow**
I'm using custom LUA scipt to do manual source switching, trying to mimic the resets and source switching workflow. I manually switched from GPS->EXTNAV before starting the mission. When EXTNAV resets we switch from EXTNAV->OPT_FLOW. Once EXT_NAV becomes healthy, we switch back from OPT_FLOW->EXTNAV. The aim is to automatically switch between sources when needed using LUA script. Reference: [ahrs-source.lua](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/ahrs-source.lua)

Help is much appreciated in understanding this behaviour. @rmackay9 @priseborough 

**Version**
4.1.5

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter(IRIS copter)

**Hardware type**
Gazebo SITL Ubuntu 18.04

**Logs**
[log.zip](https://github.com/ArduPilot/ardupilot/files/11650687/log.zip)

**Video**
[video link](https://mega.nz/file/xqMgGBTJ#MoXnW0qDA1VALphnXjfpnDoGXimGTAHN82A_kD_3KT0)

rmackay9, 2023-06-05:
Hi @Pallav1299,

I think a position jump is expected when switching from the optical flow source to the external nav source (assuming the external nav source is providing position estimates).  This is because optical flow only provides a velocity so the position will drift.  Once position estimates are available the EKF's estimated position will suddenly jump.  The controller's handling of these jumps should be fairly smooth though because the position controller is warned of the jump... so for example if the vehicle is in Loiter mode it should not physically move when the switch happens.

Pallav1299, 2023-06-06:
Hi @rmackay9, thanks for the response.

I would move to the latest firmware version once **4.4.0** is released. I have been extensively testing the EKF source set switching with the help of lua scripts and so far it has been working for me as expected. Lately, I have been trying to explore a few more things and testing them in SITL. One of those things is to recover from an unhealthy EXT_NAV state. To elaborate further, I am running my EXT_NAV algorithm, and switching from GPS to EXT_NAV as seen in the video above. The copter flies fine and everything seems pretty well. now, I am trying to mimic a scenario where my EXT_NAV would fail in air. To create this scenario, I deliberately fail my EXT_NAV algorithm in air and I detect that failure and move to optical_flow which is set in EK3_SRC3_ params. Meanwhile, my **EXT_NAV reinitializes to (0,0,0,0,0,0)** and is back to healthy state again. And I detect the healthy state and switch from optical_flow to EXT_NAV. These all operations are taking place in **GUIDED** mode. but, I have observed that the filter state changes to the new pose given by the reinitialised EXT_NAV and the controller tries to correct based on new estimates. Which is **not** the kind of behaviour I am looking for, I am looking for the kind of handling we do, when we switch from GPS to EXT_NAV the first time, in that case, the filter state is not changed, the copter does not move and the controllers are happy too, even if there is difference between the GPS estimate and EXT_NAV estimate. To tackle the problem of state change and drone movement because of the controllers, I thought it wouldn't be exhibited in **LOITER** mode. But if I do this switching in LOITER mode, I still see the state changing (which as you mentioned is expected) and drone movement (which as you mentioned is not expected) since the controllers are passed this information. 
So if we categorise this problem in 2 parts

- The state change of the filter when switching the sources
- The controllers corrective motion

What are your suggestions, how can we tackle the first issue to make sure, once we recover from unhealthy state it does not affect the filter state (for example, the way we handle from GPS to EXT_NAV) and the second problem where this information is correctly conveyed to the controller so there are no corrective motions and as you mentioned this should have not happened in loiter mode. Could you please help me understand why it happened for us?

rmackay9, 2023-06-07:
@Pallav1299,

I don't have a complete answer but I did watch the video.  I only saw the  vehicle move on the map once while in Loiter mode.  It's important of course to realise that what is shown on the GCS map may not reflect how the vehicle is actually moving especially when optical flow is being used.  I think when the copter's position on the map jumped this was the moment it switched from optical flow to external-nav.  This jump was basically a correction of the drift that had occurred while the vehicle was using optical flow.  The copter didn't actually physically move though.. or rather it probably drifted earlier when using optical flow.