ArduPilot/ardupilot #23976: Winch: add support for WINCH_DELIVER action

rmackay9, 2023-06-05:
[MAVLink's DO_WINCH command](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml) defines action "4" as "WINCH_DELIVER" meaning that winch is lowered until the package reaches the ground.

We could add support for this (perhaps using Payload Place) when using the [Daiwa winch](https://ardupilot.org/copter/docs/common-daiwa-winch.html) based on the reported tension suddenly reducing.  The [Daiwa winch driver is here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Winch/AP_Winch_Daiwa.h).

This enhancement request comes from this discussion.

ohitstarik, 2023-06-07:
hey @rmackay9 !

I'm interested in working on this. I'm looking around for examples now, mainly around the AP_Winch libraries to get myself situated. Seeing the implementation of the DO_WINCH control of the winch by length would be perfect. Is there anything else I should read into? Any general advice?

ohitstarik, 2023-06-15:
Hey @rmackay9 !

So i've "activated" use of the WINCH_DELIVER command so far by just making the winch unroll .5 meters every time a winch deliver command is sent by making changes in AP_Winch, AP_Mission, Mode_auto, and GCS_mavlink. For example, I can send

`message COMMAND_LONG 0 0 42600 0 0 4 0 0 0 0 0`

and the vehicle will accept the command and lower the winch by an intentional .5 meters. 

I'm going to be writing the delivery algo soon. I wanted to start a discussion about how this algorithm should run.

My idea so far was:

- Receive deliver command

- Reject command if winch not daiwa/does not have line tension telemetry, if there's no rangefinders, if line tension is below a certain threshold (to avoid strings without weights to flutter loosely around moving propellers), etc.

- Take rangefinder reading

- Check line tension to get an idea of package weight to compare to post-delivery weight, i.e. to know when the package is off of the winch hook. 

- lower package until line tension is below a certain delivery tension threshold. The mavlink definition notes that the winch slows down before touching down. Maybe create a slow down parameter so the user signifies how much to slow down by?

- retry if line length is 2 meters below rangefinder value, spool up to 2 meters above rangefinder value and lower to 2 meters below or until delivery tension threshold is met. 

- once tension below delivery threshold, reel back up

- wait until line is spooled back up and locked before continuing the mission

Also curious if rangefinder measurements are seen as necessary here or if estimated altitude is good enough. 

I'd love any input/discussion from the community about this. also open to bringing this up in the dev call if appropriate.

ohitstarik, 2023-07-13:
Hey @rmackay9,

This is my first pass at the implementation. Let me know what you think.

https://github.com/ArduPilot/ardupilot/pull/24292