ArduPilot/ardupilot #23989: Rover: position controller velocity I-term should be kept at zero in the forward-back direction

rmackay9, 2023-06-07:
As part of the Rover quiktune log analysis ([PR](https://github.com/ArduPilot/ardupilot/pull/23981), [discussion](https://discuss.ardupilot.org/t/rover-boat-quiktune-alpha-testers-wanted/101989)) @lthall recommended that we zero out the Position Controller's velocity PID I-term in the vehicle's forward-back direction.  This will help prevent I-term build-up in the velocity controller (in the direction of travel) which could also help improve cornering behaviour because currently in the case of a 90 degree corner, that forward-back I-term will become lateral I-term.

xianglunkai, 2023-06-07:
@rmackay9 
I think this is right! In my boat, I always set I is zero or a small value. I suggest use ADRC controller to replace PID for boat frame. The ADRC algorithm eliminates the adverse effects of I item and actively resists interference in water. In many of the mass production boat I have developed, ADRC works well!
https://github.com/ArduPilot/ardupilot/pull/23953

rmackay9, 2023-09-06:
This has been completed so closing.

closed 2023-09-06 by rmackay9