ArduPilot/ardupilot #23991: Setting ICE_RPM_CHAN to 1 corrupts RPM signal

JacksonUAS, 2023-06-07:
## Bug report

**Issue details**

Setting ICE_RPM_CHAN to 1 (RPM1) results in corrupting the RPM signal. Its not the signal itself, have monitored with an oscilloscope, but the Cubes/Ardupilots interpretation of the RPM. Setting to 0 results in clean RPM signal. Due to the RPM signal lurching around, in this setup Arduplane (4.3.5) then sends a signal to the starter to restart an already running engine.

**Version**
4.3.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

**Hardware type**
Cube Orange

**Logs**
Can be provided if needed.
[image]
[image]

JacksonUAS, 2023-06-07:
@tridge @proficnc - tag ;)

tridge, 2023-08-15:
this has got to be cross-talk between starter and RPM pin