ArduPilot/ardupilot #24008: AP_DDS: Implement Joystick support from ROS to RC Channel

Ryanf55, 2023-06-08:
## Feature request

**Describe the solution you'd like**

Use the ROS `sensor_msgs/msg/Joy` subscriber and hook it into ArduPilot to allow using a ROS joystick to control the vehicle. This is useful on SITL. 

**Describe alternatives you've considered**

Mavlink

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

I could use a recommendation from @peterbarker or @tridge on which libraries in ArduPilot to implement this.

srmainwaring, 2023-10-06:
@Ryanf55 - I've sketched out an initial implementation here: https://github.com/srmainwaring/ardupilot/tree/wips/wip-dds-joy

The approach lifts the code directly from `GCS_MAVLINK::handle_rc_channels_override` and assumes that joy inputs are the same as the RC overrides that would be supplied via MAVLink (for example using MAVProxy's `rc 3 1500`).

https://github.com/srmainwaring/ardupilot/blob/32f32f8951500c2477dd09dbf5e548aa9f9bbc99/libraries/AP_DDS/AP_DDS_Client.cpp#L454-L479

```bash
        const uint32_t tnow = AP_HAL::millis();

        //! @todo - to use same convention as MAVLink spec?
        // see: void GCS_MAVLINK::handle_rc_channels_override

        for (uint8_t i=0; i<MIN(8U, rx_joy_topic.axes_size); i++) {
          const uint16_t override_data = static_cast<uint16_t>(rx_joy_topic.axes[i]);

          // Per MAVLink spec a value of UINT16_MAX means to ignore this field.
          if (override_data != UINT16_MAX) {
              RC_Channels::set_override(i, override_data, tnow);
          }
        }
        for (uint8_t i=8; i<rx_joy_topic.axes_size; i++) {
            const uint16_t override_data = static_cast<uint16_t>(rx_joy_topic.axes[i]);

            // Per MAVLink spec a value of zero or UINT16_MAX means to
            // ignore this field.
            if (override_data != 0 && override_data != UINT16_MAX) {
                // per the mavlink spec, a value of UINT16_MAX-1 means
                // return the field to RC radio values:
                const uint16_t value = override_data == (UINT16_MAX-1) ? 0 : override_data;
                RC_Channels::set_override(i, value, tnow);
            }
        }
```

### Assumptions

- `sensor_msgs.msg.Joy.axes[i]` is a positive integer, assumed to be a PWM.
- `sensor_msgs.msg.Joy.buttons[i]` are ignored

### Alternatives

- Assume`sensor_msgs.msg.Joy.axes[i]` is normalised to `[-1, 1]`
- Convert normalised inputs to PWM in `AP_DDS`
- Either: assume the range maps to the default `[1100, 1900]`
- Or: scale using the `RCX_MIN`, `RCX_MAX` params

Ryanf55, 2024-10-20:
Completed by https://github.com/ArduPilot/ardupilot/pull/28145

closed 2024-10-20 by Ryanf55