ArduPilot/ardupilot #24020: External navigation measurement noise is overwritten by GPS reported accuracy in EKF

pavloblindnology, 2023-06-09:
If external navigation is used for position and/or speed together with GPS, then external navigation measurement errors are overwritten by GPS reported errors. Is it ok? AFAIU, the external navigation `if`-s should come 1st.
https://github.com/ArduPilot/ardupilot/blob/cf7b01d73a08aa78899c2031371cc8bb467e5e3f/libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp#L682
https://github.com/ArduPilot/ardupilot/blob/cf7b01d73a08aa78899c2031371cc8bb467e5e3f/libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp#L697

Also, when compiting GPS velocity variances used for data checks, vertical velocity is treated identically to horizontal velocity. I.e., `_gpsHorizVelNoise` & `gpsNEVelVarAccScale` are used instead of `_gpsVertVelNoise` & `gpsDVelVarAccScale` for vertical velocity, as it's done for variances used for fusion. Also, description mentions only horizontal velocity.
https://github.com/ArduPilot/ardupilot/blob/cf7b01d73a08aa78899c2031371cc8bb467e5e3f/libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp#L717
Mentioned this here
https://github.com/ArduPilot/ardupilot/commit/e80fb8b3fa301fb3857d3306099da7a763be99ee#commitcomment-117061127

The issues concern both EKF2 & 3.

rmackay9, 2023-09-22:
FYI @priseborough I think this probably is a real bug.

rmackay9, 2025-04-22:
This is resolved in latest (4.7.0-dev) now that [#29420](https://github.com/ArduPilot/ardupilot/pull/29420) has been merged

Thanks @pavloblindnology for the report!

closed 2025-04-22 by rmackay9