ArduPilot/ardupilot #24024: Compass is not detected on Skystar H743

noobishe, 2023-06-10:
Compass Matek M10Q-5883 is not detected on Skystar H743. 
Connected correctly, GPS is working, compass is not detected. All settings are standard, the port is correct, auto-detection is. Everything works in INAV, so the problem is in the ardupilot.
I tried with different firmware versions, the same problem.

[image]

FC - Skystar H743
GPS/Compass - Matek M10Q-5883
Ardupilot version -  any

jiri-jirus, 2023-08-04:
you have to recompile with modified HWDEF  make sure it contains line
`define HAL_I2C_INTERNAL_MASK 2`

[arducopter_with_bl.zip](https://github.com/ArduPilot/ardupilot/files/12256017/arducopter_with_bl.zip)
or here is my build. compass now works.

andyp1per, 2023-08-06:
@jiri-jirus this is already set on the bdshot version and I think he tried that

jiri-jirus, 2023-08-06:
> @jiri-jirus this is already set on the bdshot version and I think he tried that

oh I am sorry for my typo. the mask has to be set to Zero as on link below. then it will work[hwdef of non bdshot target](https://github.com/ArduPilot/ardupilot/blob/c8506ed4782f3867dc544c8dae845a976cf9b853/libraries/AP_HAL_ChibiOS/hwdef/SkystarsH7HD/hwdef.dat#L184)

jiri-jirus, 2023-08-06:
also please @noobishe correct issue name, it is Skystars, not "skystar" - noone is able to find this issue.

andyp1per, 2023-08-06:
Thanks for the report and diagnosis, I have created PR #24515 to fix this

closed 2023-08-10 by peterbarker

jiri-jirus, 2023-08-21:
please backport this patch to current stable. Skystars H7 users are not able to use Arducopter stable at all  when their compass is not working!

andyp1per, 2023-08-21:
This will be in Copter 4.4.1

mareksy12, 2023-08-28:
FW problem low brightness through osd ardupilot
https://www.youtube.com/watch?v=WiOuGGa5Nmk