ArduPilot/ardupilot #24038: AP_Camera: support the CAMERA_FOV_STATUS message

rmackay9, 2023-06-12:
The [CAMERA_FOV_STATUS](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L5839) message tells the GCS the Location (e.g. lat, lon and alt) of what the camera is looking at.

We could add support for this message by converting the [mount-poi](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/applets/mount-poi.lua) Lua script to C++.

SANTHOSH-MAMIDISETTI, 2023-06-15:
I want to work over this @rmackay9  can I be assigned this ?

rmackay9, 2023-06-15:
@SANTHOSH-MAMIDISETTI,

Sure, go for it.  In general we review and (hopefully) merge whichever PR gets raised first though.. still, I don't know of anyone else working on this issue.

rmackay9, 2024-01-16:
Support for this has been added now so closing.

closed 2024-01-16 by rmackay9