ArduPilot/ardupilot #24039: EK3_OGN_HGT_MASK bit 3 incorrect behavior

pavloblindnology, 2023-06-12:
## Bug report

**Issue details**

If 3rd bit of `EK3_OGN_HGT_MASK` is set together with 1st bit (EK3_OGN_HGT_MASK=5) and vertical position is taken from barometr (default parameters), copter climbs periodically indefinitely. Here is the the plot showing real altitude (barometer or gps) jumping at regular intervals, while the controller (and EKF) altitude remaining constant. The real alt increases as copter periodically thinks it's below desired altitude (100m) and simply turns on the throttle to keep it.
[image]

The reason is in `readGpsData` here
https://github.com/ArduPilot/ardupilot/blob/514434193d7c4dba259180db4a9aa83391c7cf01/libraries/AP_NavEKF3/AP_NavEKF3_Measurements.cpp#L693
When 3rd bit is set, Ardupilot doesn't use `ekfGpsRefHgt` to convert GPS alt to local alt. But it still updates `ekfGpsRefHgt` in `correctEkfOriginHeight`. And the innovation used in the latter is constantly non-zero.

The jumps in alt are introduced by the 5-sigma check here. If the check is removed, the alt rises smoothly and with acceleration.
https://github.com/ArduPilot/ardupilot/blob/514434193d7c4dba259180db4a9aa83391c7cf01/libraries/AP_NavEKF3/AP_NavEKF3_Measurements.cpp#L817

It seems the cure is to always use `ekfGpsRefHgt` to convert GPS alt to local alt in `readGpsData`.

**Version**
Copter-4.3.7

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
octa

**Hardware type**
SITL

**Logs**
[BIN log](https://drive.google.com/file/d/1L9tBvGNUPdPz3cMmc6Q-fqI0JJzFv-RY/view?usp=sharing)

peterbarker, 2024-05-23:
We've just merged a fix for this - thanks @pavloblindnology

closed 2024-05-23 by peterbarker

ddomit, 2024-06-10:
> We've just merged a fix for this - thanks @pavloblindnology

Thanks for this, is it already in a stable version i can test?