ArduPilot/ardupilot #24073: "uploader.py --force" failed on the Pixhawk6C

jianjianliu0918, 2023-06-15:
## Bug report

**Issue details**

Tools/scripts/uploader.py --force didn't work.
I was able to change the board ID using the "uploader.py --force " command before, but I failed to change the board ID back using the "uploader.py --force " command again.

**Version**
master &Copter-4.4.1 all tested

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
NULL

**Hardware type**
Pixhawk6C

**Logs**
_Please provide a link to any relevant logs that show the issue_
Tools/scripts/uploader.py --force build/Pixhawk6C/bin/arducopter.apj 
Loaded firmware for 38,0, size: 1750352 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Found board 2743,0 bootloader rev 5 on /dev/serial/by-id/usb-Holybro_Pixhawk6C-ODID-BL_41001F000F51303434333038-if00
WARNING: Firmware not suitable for this board (board_type=10051 (None) board_id=56 (Pixhawk6C))
FORCED WRITE, FLASHING ANYWAY!
Bootloader Protocol: 5
OTP:
  type: 
  idtype: =00
  vid: 00000000
  pid: 00000000
  coa: AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
  sn: 001f00413430510f38303334
ChipDes:
  family: STM32H743/753
  revision: V
Chip:
  STM32H74x_75x 20036450
Info:
  flash size: 1966080
  ext flash size: 0
  board_type: 10051
  board_rev: 0
Identification complete

Erase  : [====================] 100.0% (timeout: 7 seconds) 
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.

ubuntu:~/workarea/ardupilot$ mavproxy.py 
Auto-detected serial ports are:
/dev/serial/by-id/usb-Holybro_Pixhawk6C-ODID-BL_41001F000F51303434333038-if00
Connecting to /dev/serial/by-id/usb-Holybro_Pixhawk6C-ODID-BL_41001F000F51303434333038-if00
Connect /dev/serial/by-id/usb-Holybro_Pixhawk6C-ODID-BL_41001F000F51303434333038-if00 source_system=255
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/serial/by-id/usb-Holybro_Pixhawk6C-ODID-BL_41001F000F51303434333038-if00
MAV> link 1 down

magicrub, 2023-06-16:
I recently added --force to ./waf VEHICLE --upload --force via PR https://github.com/ArduPilot/ardupilot/pull/23757 and can confirm that master is now broken. Not sure what happened to uploader.py. I'm not sure if my testing with that PR was on master or an older branch, all it is doing is passing the --force flag through it so not sure how long this has been broken

magicrub, 2023-06-16:
@jianjianliu0918 your log there shows "FORCED WRITE, FLASHING ANYWAY!" and it forced an upload and did it. Are you saying when you went from 10051 back to 56 is when it didn't work?