ArduPilot/ardupilot #24136: GSC only operation now broken

Hwurzburg, 2023-06-25:
While checking what if anything needs to be updated in the wiki for your PR #23189 , I discovered that GCS only operation is now broken....if you set THR_FAILSAFE to 0 and arming checks to 65470 (although setting to 1 makes no difference, you can still arm), then it displays in the HUD "No RC receiver" even if RC Protocols is 0, but does not prevent arming....
once armed, 100% of the time you get a "Unhealthy AHRS" it drops to DCM and never recovers....if disarmed it bounces back to EKF3

I haven't regression tested to see when this happened, but this is a very important feature for many commercial users (although we warn that its prudent to actually have RC backup)...I know it was working in the past...maybe I am tagging this incorrectly

video demo using master
https://www.dropbox.com/s/zbzgw2pubf23sq1/norcmp4.mp4?dl=0

rmackay9, 2023-06-25:
I think this only affects Plane so I've changed the title and added a label.

Hwurzburg, 2023-06-26:
red herring....this was compassless on the bench...no yaw source=unhelathy AHRS

closed 2023-06-26 by Hwurzburg