ArduPilot/ardupilot #24142: Dangling pointer warning in GCC12

Ryanf55, 2023-06-27:
## Bug report

**Issue details**

GCC12 warns of dangling pointer in BMP388 driver. Talked to @peterbarker and @IamPete1. 

Recommendations:
```
PH: Looks to me that read_registers can register a callback that could fill in that buffer later.
I think just putting buf in the header would fix.
...
PB: 'though I'm also wondering if we should remove that set_register_rw_callback thing as being way too dangerous.  Relying on the compiler to catch us here is bad IMO.
```

I don't have time to fix it right now, but @tridge indicated he wanted it fixed in 4.4. 

**Version**

master on ccd38ddee0fb8105eb54ca2c230394a893f5001e (latest)

**Platform**
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**

```
../../libraries/AP_Baro/AP_Baro_BMP388.cpp: In member function ‘void AP_Baro_BMP388::timer()’:
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:148:34: warning: dangling pointer to ‘buf’ may be used [-Wdangling-pointer=]
  148 |         update_temperature((buf[6] << 16) | (buf[5] << 8) | buf[4]);
      |                             ~~~~~^
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:136:13: note: ‘buf’ declared here
  136 |     uint8_t buf[7];
      |             ^~~
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:148:51: warning: dangling pointer to ‘buf’ may be used [-Wdangling-pointer=]
  148 |         update_temperature((buf[6] << 16) | (buf[5] << 8) | buf[4]);
      |                                              ~~~~~^
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:136:13: note: ‘buf’ declared here
  136 |     uint8_t buf[7];
      |             ^~~
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:148:66: warning: dangling pointer to ‘buf’ may be used [-Wdangling-pointer=]
  148 |         update_temperature((buf[6] << 16) | (buf[5] << 8) | buf[4]);
      |                                                             ~~~~~^
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:136:13: note: ‘buf’ declared here
  136 |     uint8_t buf[7];
      |             ^~~
In member function ‘void AP_Baro_BMP388::timer()’,
    inlined from ‘static RetType Functor<RetType, Args>::method_wrapper(void*, Args ...) [with T = AP_Baro_BMP388; RetType (T::* method)(Args ...) = &AP_Baro_BMP388::timer; RetType = void; Args = {}]’ at ../../libraries/AP_HAL/utility/functor.h:88:28:
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:148:34: warning: dangling pointer to ‘buf’ may be used [-Wdangling-pointer=]
  148 |         update_temperature((buf[6] << 16) | (buf[5] << 8) | buf[4]);
      |                             ~~~~~^
../../libraries/AP_Baro/AP_Baro_BMP388.cpp: In static member function ‘static RetType Functor<RetType, Args>::method_wrapper(void*, Args ...) [with T = AP_Baro_BMP388; RetType (T::* method)(Args ...) = &AP_Baro_BMP388::timer; RetType = void; Args = {}]’:
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:136:13: note: ‘buf’ declared here
  136 |     uint8_t buf[7];
      |             ^~~
In member function ‘void AP_Baro_BMP388::timer()’,
    inlined from ‘static RetType Functor<RetType, Args>::method_wrapper(void*, Args ...) [with T = AP_Baro_BMP388; RetType (T::* method)(Args ...) = &AP_Baro_BMP388::timer; RetType = void; Args = {}]’ at ../../libraries/AP_HAL/utility/functor.h:88:28:
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:148:51: warning: dangling pointer to ‘buf’ may be used [-Wdangling-pointer=]
  148 |         update_temperature((buf[6] << 16) | (buf[5] << 8) | buf[4]);
      |                                              ~~~~~^
../../libraries/AP_Baro/AP_Baro_BMP388.cpp: In static member function ‘static RetType Functor<RetType, Args>::method_wrapper(void*, Args ...) [with T = AP_Baro_BMP388; RetType (T::* method)(Args ...) = &AP_Baro_BMP388::timer; RetType = void; Args = {}]’:
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:136:13: note: ‘buf’ declared here
  136 |     uint8_t buf[7];
      |             ^~~
In member function ‘void AP_Baro_BMP388::timer()’,
    inlined from ‘static RetType Functor<RetType, Args>::method_wrapper(void*, Args ...) [with T = AP_Baro_BMP388; RetType (T::* method)(Args ...) = &AP_Baro_BMP388::timer; RetType = void; Args = {}]’ at ../../libraries/AP_HAL/utility/functor.h:88:28:
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:148:66: warning: dangling pointer to ‘buf’ may be used [-Wdangling-pointer=]
  148 |         update_temperature((buf[6] << 16) | (buf[5] << 8) | buf[4]);
      |                                                             ~~~~~^
../../libraries/AP_Baro/AP_Baro_BMP388.cpp: In static member function ‘static RetType Functor<RetType, Args>::method_wrapper(void*, Args ...) [with T = AP_Baro_BMP388; RetType (T::* method)(Args ...) = &AP_Baro_BMP388::timer; RetType = void; Args = {}]’:
../../libraries/AP_Baro/AP_Baro_BMP388.cpp:136:13: note: ‘buf’ declared here
  136 |     uint8_t buf[7];
      |             ^~~
  ```

peterbarker, 2024-07-23:
OK, so this is definitely a compiler bug, given @brkydnc 's PR.  There's no reason that should work.

They appear to have fixed things in more recent compiler versions.

The correct fix for the affected compilers is probably to push/pop a pragma to ignore the warning on broken compilers.

I'm inclined to close this with a resolution of "Run GCC 13" or similar.  What do you think, @Ryanf55 ?

Ryanf55, 2024-07-23:
Sounds good. If anyone is bothered enough, that can add push/pop a pragra.

closed 2024-07-23 by Ryanf55