ArduPilot/ardupilot #24183: EKF3: GSF alignment cannot occur below 5m/s

rmackay9, 2023-07-01:
As reported in [this discussion](https://discuss.ardupilot.org/t/yaw-not-aligning-to-gsf-anymore-after-updating-from-4-2-3-to-4-4-0-beta1/101844/13) it appears that GSF alignment cannot occur unless the vehicle is traveling over 5m/s.

The issue appears to be that a new check was added in PR https://github.com/ArduPilot/ardupilot/pull/23515 that the vehicle is travelling above GPS_VEL_YAW_ALIGN_MIN_SPD (5m/s).

The issue can be replicated in SITL by doing the following with Copter:

- sim_vehicle.py --map --console -L CMAC2
- param set EK3_SRC1_YAW 8 (GSF)
- reboot
- ALT_HOLD
- arm throttle
- rc 3 2000 (wait 5 sec to allow the vehicle to climb)
- rc 3 1500 (to level out)
- rc 2 1400
- rc 2 1600
- rc 2 1500

At this point a message, "EKF yaw reset to GPS" should appear but it does not.  Next change GPS_VEL_YAW_ALIGN_MIN_SPD  to 1.0 and repeat the above test.

priseborough, 2023-07-01:
As suggested by @rmackay9 the way to resolve this is to use a different value of GPS_VEL_YAW_ALIGN_MIN_SPD for different vehicle types. Maybe it also needs to be smaller for copters (or a parameter) as small copters flying without a magnetometer will not want to fly at 5m/s to complete an in flight yaw alignment.

rmackay9, 2023-10-18:
This report may also be caused by this issue https://discuss.ardupilot.org/t/copter-4-4-1-released/107005/9

rmackay9, 2024-05-14:
This has been resolved in master now and the fix is included in Copter-4.5.2.

closed 2024-05-14 by rmackay9