ArduPilot/ardupilot #24216: AP_ Torqeedo: Supports multiple instances

xianglunkai, 2023-07-04:
I plan to expand the drive of multiple motors and need some prompts.

1. Can we refer to AP_ Rangefinder_ Backend_ Create a base class for multiple RS485 instances using serial；

2. Inherit existing drivers from xx_ Backend_ Serial;

This solution should be feasible, but there is a question: the current driver command comes from kthrottle. If there are multiple commands, how should we handle them：Using kthrottle to unify left/right? Should further separation be necessary?

Thank you very much for the discussion!

rmackay9, 2023-07-04:
Hi @xianglunkai,

Yes, we should split the AP_Torqeedo library into a frontend and backend similar to what we've done for other libraries including the AP_Range library.

I think we will need a new parameter to specify either which servo output channel to use (as an input) or to specify the function that should be used.  I was originally planning to use the function... many of our ESC output libraries (e.g. KDECAN, PiccoloCAN) drivers use a servo output mapping though.   I think using the function is best though.

xianglunkai, 2023-07-04:
@rmackay9 
Thank you very much!
I will try solve it recently.

rmackay9, 2023-07-04:
@xianglunkai,

Thanks, this is really great news.  By the way, based on top of this, I'd like to add support for the ePropulsion motors ([WIP branch here](https://github.com/rmackay9/rmackay9-ardupilot/tree/epropulsion1)) and also the higher powered Torqeedo motors that use J1939 CAN protocol (see "Nubbish" in the Discord canbus disucussion).

There's a related discussion here https://discuss.ardupilot.org/t/differential-steering-w-torqeedo-motors/78823

rmackay9, 2023-08-03:
[Here's another related discussion](https://discuss.ardupilot.org/t/torqeedo-cruise-3-0-fp-fixed-pod-compatibility/104490) from a user who would like to use the more powerful Torqeedo motors with AP.  As mentioned above these motors use the J1939 CAN or maybe CAN2000 protocol (maybe they are the same thing).

I have also put a copy of Nubbish's Lua script here: https://www.dropbox.com/scl/fi/azgidbnmw3a3fsptrmvm2/CAN_TorqLink.lua?rlkey=liw0un5e1q3kjnvdo2x7976sc&dl=0

xianglunkai, 2023-08-03:
@rmackay9 
OK！Thank you very much!
I will complete it as soon as possible. Recently, the project has been a bit busy and needs to be delayed for some time

rmackay9, 2024-05-08:
This has been resovled with PR https://github.com/ArduPilot/ardupilot/pull/26494 getting merged to master.  This change will be include in AP 4.6.