ArduPilot/ardupilot #24235: There may be a bug in void AP_MotorsMatrix::output_armed_stabilizing()

luweiagi, 2023-07-06:
@lthall 
bug is in `void AP_MotorsMatrix::output_armed_stabilizing()` at
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsMatrix.cpp#L366

code should be modified from A to B:
A:
[image]

B:
[image]

rmackay9, 2023-07-06:
@luweiagi,

Thanks for the investigation and report.

Maybe you could provide some more evidence showing the current values are incorrect?  Perhaps pass in some test value and provide the resulting output.  Another alternative may be to test in SITL to see if you can recreate situations where the existing formula comes up with the incorrect output?

Sorry to be annoying but it takes quite a lot of effort to investigate these reports and most of the time we determine that there is no actual error.  It's more often something has been overlooked in the report.

luweiagi, 2023-07-06:
> @luweiagi,
>
> Thanks for the investigation and report.
>
> Maybe you could provide some more evidence showing the current values are incorrect? Perhaps pass in some test value and provide the resulting [...]

OK, I have updated it.

lthall, 2023-07-06:
Hi @luweiagi,

As always, I appreciate you looking so closely at the code!!

I don't believe this is a bug. The purpose of rpy_scale is to limit the roll, pitch and yaw if we want to prioritise low throttle. When we have high throttle we always prioritise roll, pitch and yaw over throttle. This means we only adjust the rpy_scale if the lower limit can't fit. If the upper limit can't fit then we reduce the throttle to give it room.

The second limit problem you pointed out isn't really a bug but more of an approximation. It is possible that one of the three axis may be able to be increased without increasing the saturation but the calculations required to remove this approximation would be expansive and complicated.

Have I missed something?

luweiagi, 2023-07-10:
> Hi @luweiagi,
>
> As always, I appreciate you looking so closely at the code!!
>
> I don't believe this is a bug. The purpose of rpy_scale is to limit the roll, pitch and yaw if we want to prioritise low [...]

Thanks for your patient reply! 
I get your meaning of " The purpose of rpy_scale is to limit the roll, pitch and yaw if we want to prioritise low throttle. When we have high throttle we always prioritise roll, pitch and yaw over throttle. This means we only adjust the rpy_scale if the lower limit can't fit. If the upper limit can't fit then we reduce the throttle to give it room."

> The second limit problem you pointed out isn't really a bug but more of an approximation. It is possible that one of the three axis may be able to be increased without increasing the saturation but the calculations required to remove this approximation would be expansive and complicated.

I have a case I can not figure out:
case: avg_max=0.3, rpy_low=-0.2, rpy_high=1.8, throttle_thrust=0.2.
If we go with the logic of the second bug, we can get:
[image]

peterbarker, 2024-11-25:
@lthall ball is in your court on this one, I think.

lthall, 2025-05-19:
Sorry @luweiagi, I missed your follow up. I think the problem appears here because you are using values of rpy_low and rpy_high that we can't obtain with a real motor mixer. The key limitation is that the inputs roll, pitch, yaw and thrust are orthogonal. 

If we have roll, pitch and yaw all one we would get quad X outputs of 3, -1, -1, -1. 
```
    // calculate any scaling needed to make the combined thrust outputs fit within the output range
    float rpy_scale = 1.0f;
    if (rpy_high - rpy_low > 1.0f) {
        rpy_scale = 1.0f / (rpy_high - rpy_low);
    }
```
rpy_scale = 1/4 = 0.25
```
    if (throttle_avg_max + rpy_low < 0) {
        rpy_scale = MIN(rpy_scale, -throttle_avg_max / rpy_low);
    }
```
if throttle_avg_max = 0.3, throttle_avg_max + rpy_low < 0
-throttle_avg_max / rpy_low = 0.3 > 0.25
```
    // calculate how close the motors can come to the desired throttle
    rpy_high *= rpy_scale;
    rpy_low *= rpy_scale;
```
We scale that down to get 0.75, -0.25, -0.25, -0.25
rpy_high = 0.75,
rpy_low = -0.25
```
    throttle_thrust_best_rpy = -rpy_low;
```
throttle_thrust_best_rpy  = - rpy_low = 0.25
```
    float thr_adj = throttle_thrust - throttle_thrust_best_rpy;
```
throttle_thrust = 0.2
thr_adj  = 0.2 - 0.25 = -0.05
```
    if (rpy_scale < 1.0f) {
        // Full range is being used by roll, pitch, and yaw.
        limit.roll = true;
        limit.pitch = true;
        limit.yaw = true;
        if (thr_adj > 0.0f) {
            limit.throttle_upper = true;
        }
```
thr_adj = -0.05 < 0.0
```
        thr_adj = 0.0f;
    } else if (thr_adj < 0.0f) {
        // Throttle can't be reduced to desired value
        // todo: add lower limit flag and ensure it is handled correctly in altitude controller
        thr_adj = 0.0f;
    } else if (thr_adj > 1.0f - (throttle_thrust_best_rpy + rpy_high)) {
        // Throttle can't be increased to desired value
        thr_adj = 1.0f - (throttle_thrust_best_rpy + rpy_high);
        limit.throttle_upper = true;
    }
```

So I think the reason you are getting this funny result is you are using values that we can't get with a properly orthogonal mixer.