ArduPilot/ardupilot #24245: SITL: dumpstack/dumpcore do not run if PWD is not in the repository and sh != bash

durka, 2023-07-07:
## Bug report

**Issue details**

There is a comment here: https://github.com/ArduPilot/ardupilot/blob/a480c0a74a3edc7e9905bfd9821dfa973d416077/libraries/AP_HAL_SITL/system.cpp#L58

Unfortunately, trusting in `$PATH` is a hazardous course of action! This function uses `sh` to invoke the program. On my system (Ubuntu 18.04), `sh` is symlinked to `dash`, which does NOT search `$PATH` when given a command that doesn't include a slash. 

The man page makes it sound like it does, but it doesn't.

>     Path Search
>      When locating a command, the shell first looks to see if it has a shell function by that name.  Then it looks for a builtin command by that name.  If a builtin command is not found, one of two things happen:
>      1.   Command names containing a slash are simply executed without performing any searches.
>      2.   The shell searches each entry in PATH in turn for the command.  The value of the PATH variable should be a series of entries separated by colons.  Each entry consists of a directory name.  The current directory may be indicated implicitly by an empty directory name, or explicitly by a single period.

```
$ mkdir test
$ echo -e '#!/bin/sh\necho $0 "$@"\n' >test/test.sh
$ PATH=$PWD/test:$PATH bash test.sh 1 2 3
test.sh 1 2 3
$ PATH=$PWD/test:$PATH dash test.sh 1 2 3
dash: 0: cannot open test.sh: No such file
```

Here are two workarounds:
1. Use `sh -c`, but note that in that case the full command needs to be in quotes, i.e. `sh -c "dumpstack.sh PID"` or `sh -c "dumpstack.sh $0" PID`
2. If the command isn't found in one of the canned search directories, find it from `$PATH` using `which`

**Version**
4.2.3

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
[x] SITL

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

peterbarker, 2024-08-03:
We can remove any hack required for this when this sequence of commands results in something sensible:

```
pbarker@fx:~/rc/ardupilot(master)$ ls ~/bin/bin2elf.sh 
/home/pbarker/bin/bin2elf.sh
pbarker@fx:~/rc/ardupilot(master)$ bash bin2elf.sh
Usage: bin2elf.sh BINFILE ELFFILE ADDRESS
pbarker@fx:~/rc/ardupilot(master)$ dash bin2elf.sh
dash: 0: cannot open bin2elf.sh: No such file
pbarker@fx:~/rc/ardupilot(master)$ 
```

peterbarker, 2024-08-03:
@durka OK, you've obviously got a slightly weird setup where it can't find the thing in the path it is expecting, but that you *are* adding that scripts directory to the path.

I'm uncomfortable changing that code for everyone, but please see the PR I just created (https://github.com/ArduPilot/ardupilot/pull/27742) to give people some sort of an option, at least.

Look reasonable to you?

durka, 2024-08-05:
I'll comment over in there, we can close this if it's not the desired approach

closed 2024-08-07 by tridge