ArduPilot/ardupilot #24272: AP_GPS: DroneCAN GPS_AUTOCONFIG does not revert from moving baseline to standard

yuri-rage, 2023-07-12:
## Bug report

**Issue details**
When using AP_Periph GPS modules (Here3+) with GPS_AUTOCONFIG=2, GPS_TYPE=22, and GPS_TYPE2=23, ArduPilot will correctly set the periph node GPS types to 17 and 18.

However, if the user subsequently reverts to GPS_TYPE=9, ArduPilot will fail to detect the periph GPS until it is manually set back to GPS_TYPE=1 within the CAN parameter list.

Discuss topic:
https://discuss.ardupilot.org/t/here3-issues-after-failed-gps-yaw-attempt/103589/

**Version**
Rover 4.2.3, Copter 4.3.7

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[ x ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
Cube Orange, Cube Orange+, Here3+

**Logs**
N/A

ppastor304, 2023-07-24:
Same problem and same solution with Plane V4.3.7

yuri-rage, 2023-10-11:
Further discussion seems to indicate that Here model DroneCAN modules do not function properly at all in moving baseline configurations.

https://discuss.ardupilot.org/t/herepro-moving-baseline-configuration-issues/107156/

yuri-rage, 2023-10-16:
Seems I had a bit of a misconception regarding the available feature set on the Here3+. They are not compatible with a moving baseline config, which probably explains some of this.

However, HerePro modules may be similarly afflicted - request this issue be kept open until it can be confirmed that HerePro modules function properly with a moving baseline configuration. At present, it seems they do not.