ArduPilot/ardupilot #24305: Doc: will AP_Mission_test make the drone actual fly, or it’s just simulation? if actual fly, need to add warning doc

mw66, 2023-07-14:
Hi,

I asked my question here, but seems nobody knows the answer:

https://discuss.ardupilot.org/t/newbie-q-will-ap-mission-test-make-the-drone-actual-fly-or-its-just-simulation/103686/4

## Bug report

**Issue details**

Hi,

If I build this test and upload to my drone, will it actually fly? Or the test run is just a simulation?

Sorry for the silly newbie question, since I am study the material indoor, I really do not want to see my drone hit something indoor.

And if it make the drone actually fly, how much time I have after the upload is done and before the drone take off? At least I need time to unplug the USB cable.

I know MissionPlanner. But I am more interested in writing direct cpp code to achieve a mission (for a very specific purpose).

That is why I want to try that code, hence the question.

Thanks.

https://github.com/ArduPilot/ardupilot/blob/ec994957b3b2a16e8347f22d9804a96103944aa0/libraries/AP_Mission/examples/AP_Mission_test/AP_Mission_test.cpp#L184

This line is especially dangerous, since it specifies exact long & lat, if this code will make the drone actually fly, we need to add warning doc to the code, to let the users know.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2023-07-14:
This code will not cause the vehicle to fly.

closed 2023-07-14 by IamPete1

mw66, 2023-07-14:
> This code will not cause the vehicle to fly.

@IamPete1 

Thanks. Can we add to the doc as comments?

BTW, if I do *want* to code the mission this way in C++, how to make the drone actually fly? any sample code elsewhere?

Thanks.