ArduPilot/ardupilot #24306: AP_DDS: Protect subscribers from crashing the main thread

Ryanf55, 2023-07-14:
## Bug report

**Issue details**

In AP_DDS subscribers (joystick or tf), the Autopilot is forced to subscribe and parse incoming messages regardless of size or frequency. It has been observed on current `master` branch that sufficiently large joystick array sizes immediately cause a segfault. Although DDS runs on its own thread, a segfault in AP causes the main thread to crash. Tagging @srmainwaring for raising this issue.

Because ArduPilot has no control over how fast messages will arrive, and DDS is treated as a public API, with potentially extreme data rates over UDP (ethernet), this is a large vulnerability. 

**Version**

Observed on `master` July 14th ec994957b3b2a16e8347f22d9804a96103944aa0

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
SITL, Pixhawk 6X

**Logs**
TBD

**Actions** 

* Determine what protections can be done
* Find root cause on the segfault
* Define a plan for preventing excessive CPU and RAM usage from AP_DDS - can we ignore samples or slow the reading down if memory limits are getting close? 
* Implement the protection

This is tagged for **4.5**; it must be done by then otherwise DDS should not be enabled by default on any platform except SITL.

srmainwaring, 2023-07-14:
Additional info:

- The segfault occurred while testing https://github.com/ArduPilot/ardupilot/pull/24155 and was observed subscribing to tf2_msgs/msg/TFMessage, not sensor_msgs/msg/Joy, although the same issue may be present for the latter.
- The frequency of data does not appear to be an issue, or at least has not caused a segfault to date. Publishing smaller TF payloads at high frequency with
  
  ```bash
    ros2 topic pub /ap/tf tf2_msgs/msg/TFMessage 'transforms: [
    {header: {stamp: {sec: 227, nanosec: 91000000}, frame_id: iris}, child_frame_id: iris/tilt_link, transform: {translation: {x: 0.019984060916003184, y: -0.020015926647110993, z: -0.09998399286175172}, rotation: {x: -0.4999997609802308, y: 0.49999975742835057, z: 0.4996019204931798, w: 0.5003982440325244}}}]' --rate 1000
  ``` 

  for example does not cause an issue, although the throughput to AP_DDS is significantly lower than the desired rate.

Examples  to replicate here: https://github.com/ArduPilot/ardupilot/pull/24155#issuecomment-1635795311, with further details to follow.

srmainwaring, 2023-07-21:
Closed by https://github.com/ArduPilot/ardupilot/pull/24348

closed 2023-07-21 by srmainwaring