ArduPilot/ardupilot #24319: Issues related to formula in AP_L1_Control::nav_roll_cd

abaghiyan, 2023-07-15:
## Bug report

**Issue details**

In method [AP_L1_Control::nav_roll_cd](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_L1_Control/AP_L1_Control.cpp#L79) used formula `ret = cosf(_ahrs.pitch)*degrees(atanf(_latAccDem * (1.0f/GRAVITY_MSS)) * 100.0f);` has a hard to understand physical meaning. Shouldn't it be as follows:
`ret = degrees(atan2f(_latAccDem, GRAVITY_MSS * cosf(_ahrs.pitch)))*100.0f`, as for balanced spiral flight (pitching and rolling at the same time) the equations of motion can be written as follows:
`liftForce * cos(roll) = gravityForce * cos(pitch);`
`liftForce * sin(roll) = gravityForce * lateralAcceleration / gravityAcceleration; // as gravityForce/gravityAcceleration = mass `   
From the above equations the desired roll can be adapted to the proposed code.

**Version**
4.3.7

**Platform**
Plane

**Airframe type**
Plane airframes

**Hardware type**
Pixhawk

IamPete1, 2023-07-16:
I think your right, this is what the two equations look like:

[image]

I rewrote:
```
atan2(latAccDem, GRAVITY_MSS * cos(pitch)))
```
to:
```
atan(latAccDem * (1.0/(GRAVITY_MSS * cos(pitch))))
```
The result is the same and it makes it a little easier to relate back to the current equation.

Currently we reduce roll an high/low pitch we should be increasing it to make up for the reduced lift required.

Can you do a PR?

abaghiyan, 2023-07-16:
> I think your right, this is what the two equations look like:
>
> [image]
>
> I rewrote:
>
> ```
> atan2(latAccDem, GRAVITY_MSS * cos(pitch)))
> ```
>
> to:
>
> ```
> atan(latAccDem * (1.0/(GRAVITY_MSS * cos(pitch)))) [...]

Dear @IamPete1, I think I can do a PR to fix the issue.