ArduPilot/ardupilot #24320: Issues related to formula in Plane::update_load_factor

abaghiyan, 2023-07-15:
## Bug report

**Issue details**

In method [Plane::update_load_factor](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/Attitude.cpp#L582) to estimate aerodynamic load factor the following formula is used `aerodynamic_load_factor = 1.0f / safe_sqrt(cosf(radians(demanded_roll)));`. In general when there the pitch and roll exist the formula for the aerodynamic load factor should be as follows: `aerodynamic_load_factor = cosf(radians(_ahrs.pitch))/cosf(radians(demanded_roll))`, which in case of turns (in horizontal plane) becomes `aerodynamic_load_factor = 1.0f/cosf(radians(demanded_roll))`. Why square root is used in the existing code. If proposed formulas are accepted shouldn't the code `int32_t roll_limit = degrees(acosf(sq(1.0f / max_load_factor)))*100;` also be changed accordingly (without square and taking into account pitch).

**Version**
4.3.7

**Platform**
Plane

**Airframe type**
Plane airframes

**Hardware type**
Pixhawk

IamPete1, 2023-07-16:
I think all our maths give the correct results, just that were just doing the sqrt in the wrong spot?

I think we would accept a PR to fix it, remove the sqrt from the load factor and the square from the roll limit cal. We would also need to add a sqrt here in that case:

https://github.com/ArduPilot/ardupilot/blob/bde43f167ef535c1c301c90528d91b4fa0815a91/libraries/AP_TECS/AP_TECS.cpp#L418

Is that something you could do?

Once that is corrected we could think about including the pitch. That will reduce the load factor, so we might be better to ignore to stay on the conservative side.

abaghiyan, 2023-07-16:
> I think all our maths give the correct results, just that were just doing the sqrt in the wrong spot?
>
> I think we would accept a PR to fix it, remove the sqrt from the load factor and the square [...]

Dear @IamPete1, thank you for the quick response, I think I can do the changes and create a PR to fix the issue (will leave the pitch part for future).

abaghiyan, 2023-08-02:
Dear Peter Hall (@IamPete1) the connection between minimum airspeed and load factor can be obtained through equation:
`liftForce = 0.5 * liftCoeff * rho * v_min * v_min * area = gravityForce //(in case of loadFactor = 1)`
`v_min = sqrt(2*gravityForce/(liftCoeff * rho * area))`
For cases when `loadfactor ≠ 1` we have the following formula:
`liftForce = 0.5 * liftCoeff * rho * v_min * v_min * area = loadFactor * gravityForce `
`v_minwithloadFacor = sqrt(2 * loadFactor * gravityForce/(liftCoeff * rho * area)) = sqrt(loadFacor) * v_min` which should be in [ardupilot/libraries/AP_TECS/AP_TECS.cpp](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.cpp#L418). 
To connect load factor to speed ratio between smoothed airspeed `v_smooth` and minimum airspeed `v_min` the last formula can be used for `v_smooth`:
`v_smooth = sqrt(loadFactor) * v_min`, from which
`loadFactor = square(v_smooth/v_min`, so formula in [Plane::update_load_factor](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/Attitude.cpp#L613) should be changed to:
`float max_load_factor = sq(smoothed_airspeed / MAX(aparm.airspeed_min, 1));`

So if you approve the above I'll add new commit to PR #24469 to fix the above formula. As from tests I see that test-sitl-plane failed (AT-0153.2: Exception caught: roll-over: Failed to hold set attitude)

abaghiyan, 2023-08-05:
Added all fixes in PR #24509 . All tests passed now.