ArduPilot/ardupilot #24330: how can H750 support compiling complete firmware when using external storage?

suiaiyun, 2023-07-17:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Hello, how can H750 support compiling complete firmware when using external storage? FLASH_ SIZE_ When KB is set to 128, the compiled firmware is not complete. What can I do to obtain complete firmware?

andyp1per, 2023-07-17:
This is a support request - closing.

Please look at one of the boards that does this e.g. SPRacingH7

closed 2023-07-17 by andyp1per

suiaiyun, 2023-07-18:
> 这是一个支持请求 - 关闭。
> 
> 请查看执行此操作的主板之一，例如 SPRacingH7

Perhaps my description was not detailed enough, let me rephrase the problem I encountered:

1. When 'FLASH_ SIZE_ KB 128' is configured in hwdef.dat, the compiled firmware does not include Script function, and SRC_ENABLE cannot be found in the parameter list_ ENABLE parameter.

2. When 'define HAL_USE_EMPTY_STORAGE 1' is configured in hwdef.dat, all modifications made to the parameters will not be saved and will be restored to their original state after restarting. If this line is not added, the parameter is saved in a file called arduplane. stg on the SD card, and if the SD card flight control is not installed, it cannot run.

Thank you to every developer for their efforts.

andyp1per, 2023-07-20:
1. Can you force scripting enable in the hwdef?
2. Yes, you have to use SD card for parameter storage