ArduPilot/ardupilot #24334: FCU communication via UDP port not working

oselin, 2023-07-17:
## Bug report
In the latest version of Copter (4.5.0) the FCU communication via UDP port is not working, if trying to communicate via Mavlink (MAVROS)
**Issue details**
A ROS2 node, initialized in python as following
```python3
mavros_node = Node(
        package='mavros',
        executable= 'mavros_node',
        parameters=[
            {'fcu_url': 'udp://127.0.0.1:14551@14555'},
            {'gcs_url': ''},
            {'target_system_id': 1},
            {'target_component_id': 1},
            {'fcu_protocol': 'v2.0'},
            apm_pluginlists_file,
            apm_config_file,
        ],

        namespace='/drone1',
        output='screen',
        respawn=True,
)
```
I had to downgrade to version 4.3.0.
**Version**
Copter-4.5.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
gazebo-iris

khancyr, 2023-07-17:
How do you launch SITL ?

oselin, 2023-07-24:
The SITL is launched by using the following bash script
https://github.com/oselin/drone-pose-estimation/blob/main/src/iq_sim/scripts/simulation.sh

while the environment was set up by using
https://github.com/oselin/drone-pose-estimation/blob/main/install.sh

I basically followed the the following tutorials
https://www.youtube.com/watch?v=r15Tc6e2K7Y&list=PLy9nLDKxDN6-34W07yZECzp_Yc2WmWIYH&index=1

https://www.youtube.com/watch?v=kcCL0w4NbIc&list=PLy9nLDKxDN6-34W07yZECzp_Yc2WmWIYH&index=3

https://github.com/Intelligent-Quads/iq_tutorials/tree/master/docs

By simply downgrading to ArduCopter v4.3.0 or v4.3.1 it works. I fixed the issue in this way

farhan-haroon, 2023-07-28:
Hey 
Are using MAVROS with MAVPROXY simulteneously?
If so, how?
Can I connect with my physical quadcopter and mavproxy and mavros using the same telemetry radio?