ArduPilot/ardupilot #24358: STM32H750VB uses external flash to store parameters?

suiaiyun, 2023-07-20:
**Issue details**

Hello, I am currently migrating flight control based on STM32H750VBT6 master control. I have used w25q128 as external storage and the code version is Plane-4.4 beta3. The current issues are as follows:

1. When "FLASH_SIZE_KB 128" is set in the hwdef.dat file, the resulting firmware does not have the parameter "SCR ENABLE", and the script function cannot be used. When set to "FLASH_SIZE_KB 2048", I gained script functionality and was able to run normally. Is there a hidden danger in my doing this?

2. Can parameters only be saved in memory cards? Is there a way to save to external storage? Because FC cannot function properly without installing a memory card.

Thank you to every developer for their work.

**Version**
4.4-beta3

**Platform**
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

andyp1per, 2023-07-20:
Yes you need to use SD card to store parameters.

Looks like the scripting setup is hardcoded to the flash size, probably needs a fix.

andyp1per, 2023-07-20:
You could try setting
```
define AP_SCRIPTING_ENABLED 1
define LUA_32BITS 1
```
in hwdef

suiaiyun, 2023-07-21:
> Yes you need to use SD card to store parameters.
OK. Do you have any plans to change it in the future and save the parameters in external storage?

> Looks like the scripting setup is hardcoded to the flash size, probably needs a fix.

When I set
```bash
FLASH_SIZE_KB 2048
FLASH_RESERVE_START_KB 0

EXT_FLASH_SIZE_MB 8
EXT_FLASH_RESERVE_START_KB 1024
EXT_FLASH_RESERVE_END_KB 448
```
I obtained the following firmware:
```bash
Target         Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
--------------------------------------------------------------------------------
bin/arduplane   1731136      3432   267636               1734568         5146700
```

When I set
```bash
FLASH_SIZE_KB 128
FLASH_RESERVE_START_KB 0

EXT_FLASH_SIZE_MB 8
EXT_FLASH_RESERVE_START_KB 1024
EXT_FLASH_RESERVE_END_KB 448
```
I obtained the following firmware:
```bash
Target         Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
--------------------------------------------------------------------------------
bin/arduplane   1319408      2564   268484               1321972         5559308
```
Firmware size seems to be influenced by internal storage size. and for FCs using external storage, firmware size should not be affected by internal storage.

suiaiyun, 2023-07-21:
> 

Okay, I will try this method.

suiaiyun, 2023-07-21:
> You could try setting
> 
> ```
> define AP_SCRIPTING_ENABLED 1
> define LUA_32BITS 1
> ```
> 
> in hwdef

Okay, I will try this method.

andyp1per, 2023-07-21:
> Good. Do you have any plans to change it in the future and save the parameters in external storage?

I started some time back - https://github.com/ArduPilot/ardupilot/pull/20210 - but have not had the incentive to finish it. It's not that easy.

suiaiyun, 2023-07-21:
> I started some time back - #20210 - but have not had the incentive to finish it. It's not that easy.

Great, thank you for your work. This is a highly anticipated feature.

andyp1per, 2023-07-21:
> Great, thank you for your work. This is a highly anticipated feature.

It's unlikely that I will work on it anytime soon

suiaiyun, 2023-07-21:
> It's unlikely that I will work on it anytime soon

From the current situation, saving parameters in a memory card is also a good choice, at least allowing STM32H750VB to run. I will continue testing it, it is a good choice.

andyp1per, 2023-07-21:
> From the current situation, saving parameters in a memory card is also a good choice, at least allowing STM32H750VB to run. I will continue testing it, it is a good choice.

Yeah it generally works fine - I have not had any issues with it

suiaiyun, 2023-07-27:
> > From the current situation, saving parameters in a memory card is also a good choice, at least allowing STM32H750VB to run. I will continue testing it, it is a good choice.
> 
> Yeah it generally works fine - I have not had any issues with it

Hi, andyp1per. I have discovered a new issue. After setting many waypoint in my flight plan and uploading it to flight control, I soon received a notification message:
```bash
2023-07-27 14:52:48 : IE0 IEC0 TN:
2023-07-27 14:52:48 : WDG: T-3 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0 
2023-07-27 14:52:38 : IE0 IEC0 TN:
2023-07-27 14:52:38 : WDG: T-3 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0 
2023-07-27 14:52:28 : IE0 IEC0 TN:
2023-07-27 14:52:28 : WDG: T-3 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0 
2023-07-27 14:52:18 : IE0 IEC0 TN:
2023-07-27 14:52:18 : WDG: T-3 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0 
2023-07-27 14:52:08 : IE0 IEC0 TN:
2023-07-27 14:52:08 : WDG: T-3 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0 
2023-07-27 14:51:58 : IE0 IEC0 TN:
2023-07-27 14:51:58 : WDG: T-3 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0 
2023-07-27 14:51:48 : IE0 IEC0 TN:
2023-07-27 14:51:48 : WDG: T-3 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0 
```
This is a problem discovered during static testing, and it does not occur without a planned waypoint. It is unclear whether it will affect actual flight, but now it seems that the flight control attitude response is normal.

andyp1per, 2023-07-27:
A watchdog is never good. You should try attaching a debugger and see if you can see what is causing it.

suiaiyun, 2023-07-27:
> A watchdog is never good. You should try attaching a debugger and see if you can see what is causing it.

I found that deleting all waypoints is normal, and if there are waypoints in the flight control, this problem will occur.